robotology/mesh-support #1901: Right shoulder roll encoder AEA invalid data

mebbaid, 2024-09-12:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Upon starting the robot interface with @vigisushrutha23  for a late experiment the arm did not complete the calibration due to a hardware fault with the message AEA encoder invalid data

### Detailed context

Upon inspection, I found this wire cut (not sure its related to the AEA encoder message) 

[image]

### How does it affect you?

This affects ICRA deadline experiments

AntonioAzocar, 2024-09-16:
I've repaired the AMO cable, so then please give me feedback!

cc @AntonioConsilvio

S-Dafarra, 2024-09-17:
The calibration does not go well. The joint goes to the opposite limit

vigisushrutha23, 2024-09-20:
We were able to run experiments yesterday with no issues, but arm FTs were disabled. I think we can close this issue as another issue has been #1909 has been opened about the FTs.

AntonioConsilvio, 2025-01-27:
Hi! we fixed this issue. The problem was the SPI connector on the EMS board (connector P8). We changed the connection to the P10 connector and it works properly now. Closing ✅