robotology/mesh-support #1903: Problem with external fault board

FabioBergonti, 2024-09-13:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The robot does not start

### Detailed context

We are trying to start the robot but the yarprobotinterface fails with these errors:
```
	41,661508	ERROR			from BOARD 10.0.1.1 (left_arm-eb1-j0_1) time=20s 526m 26u :  MC: external fault button pressed.
	41,671551	ERROR			from BOARD 10.0.1.1 (left_arm-eb1-j0_1) time=20s 527m 52u :  MC: external fault button pressed.
	41,773185	ERROR			from BOARD 10.0.1.2 (left_arm-eb2-j2_3) time=20s 646m 355u :  MC: external fault button pressed.
	41,773191	ERROR			from BOARD 10.0.1.2 (left_arm-eb2-j2_3) time=20s 646m 480u :  MC: external fault button pressed.
	41,884793	ERROR			from BOARD 10.0.1.3 (right_arm-eb3-j0_1) time=20s 737m 208u :  MC: external fault button pressed.
	41,884798	ERROR			from BOARD 10.0.1.3 (right_arm-eb3-j0_1) time=20s 737m 334u :  MC: external fault button pressed.
	41,996215	ERROR			from BOARD 10.0.1.4 (right_arm-eb4-j2_3) time=20s 846m 307u :  MC: external fault button pressed.
	41,996221	ERROR			from BOARD 10.0.1.4 (right_arm-eb4-j2_3) time=20s 846m 432u :  MC: external fault button pressed.
	42,138225	ERROR			from BOARD 10.0.1.7 (left_leg-eb7-j0_2) time=20s 990m 320u :  MC: external fault button pressed.
	42,138231	ERROR			from BOARD 10.0.1.7 (left_leg-eb7-j0_2) time=20s 991m 349u :  MC: external fault button pressed.
	42,138236	ERROR			from BOARD 10.0.1.7 (left_leg-eb7-j0_2) time=20s 991m 478u :  MC: external fault button pressed.
	42,280098	ERROR			from BOARD 10.0.1.8 (left_leg-eb8-j3_5) time=21s 123m 692u :  MC: external fault button pressed.
	42,280123	ERROR			from BOARD 10.0.1.8 (left_leg-eb8-j3_5) time=21s 124m 670u :  MC: external fault button pressed.
	42,280130	ERROR			from BOARD 10.0.1.8 (left_leg-eb8-j3_5) time=21s 124m 824u :  MC: external fault button pressed.
	42,411998	ERROR			from BOARD 10.0.1.11 (right_leg-eb11-j0_2) time=21s 280m 240u :  MC: external fault button pressed.
	42,422047	ERROR			from BOARD 10.0.1.11 (right_leg-eb11-j0_2) time=21s 281m 268u :  MC: external fault button pressed.
	42,422054	ERROR			from BOARD 10.0.1.11 (right_leg-eb11-j0_2) time=21s 281m 397u :  MC: external fault button pressed.
	42,533506	ERROR			from BOARD 10.0.1.12 (right_leg-eb12-j3_5) time=21s 400m 877u :  MC: external fault button pressed.
	42,533523	ERROR			from BOARD 10.0.1.12 (right_leg-eb12-j3_5) time=21s 401m 855u :  MC: external fault button pressed.
	42,543622	ERROR			from BOARD 10.0.1.12 (right_leg-eb12-j3_5) time=21s 402m 15u :  MC: external fault button pressed.
	42,645276	ERROR			from BOARD 10.0.1.20 (head-eb20-j0_1) time=22s 153m 154u :  MC: external fault button pressed.
	42,645283	ERROR			from BOARD 10.0.1.20 (head-eb20-j0_1) time=22s 153m 258u :  MC: external fault button pressed.
	42,797619	ERROR			from BOARD 10.0.1.21 (head-eb21-j2_5) time=22s 301m 633u :  MC: external fault button pressed.
	42,797626	ERROR			from BOARD 10.0.1.21 (head-eb21-j2_5) time=22s 301m 747u :  MC: external fault button pressed.
	42,797636	ERROR			from BOARD 10.0.1.21 (head-eb21-j2_5) time=22s 301m 856u :  MC: external fault button pressed.
	42,807733	ERROR			from BOARD 10.0.1.21 (head-eb21-j2_5) time=22s 301m 960u :  MC: external fault button pressed.
	42,878781	ERROR			from BOARD 10.0.1.22 (face-eb22-j0) time=22s 387m 587u :  MC: external fault button pressed.
	42,970271	ERROR			from BOARD 10.0.1.5 (torso-eb5-j0_2) time=21s 897m 789u :  MC: external fault button pressed.
	42,970277	ERROR			from BOARD 10.0.1.5 (torso-eb5-j0_2) time=21s 898m 790u :  MC: external fault button pressed.
	42,970282	ERROR			from BOARD 10.0.1.5 (torso-eb5-j0_2) time=21s 898m 909u :  MC: external fault button pressed.

``` 

We tried multiple times with different fault buttons, but the errors persit

### Additional context

[log_icub-head_yarprobotinterface[robotinterface]_2797.log](https://github.com/user-attachments/files/16992838/log_icub-head_yarprobotinterface.robotinterface._2797.log)

### How does it affect you?

We cannot do experiments

AntonioAzocar, 2024-09-13:
I found the fault connector of the FED damaged (as you can see in the video) and it wasn't repairable, so I replaced the board directly. 

https://github.com/user-attachments/assets/aa7857a4-a718-4f84-bc57-79ca4b1f9d68

cc @AntonioConsilvio

AntonioConsilvio, 2024-09-16:
HI! Good job @AntonioAzocar!

@FabioBergonti any feedback on this? Thank you!

cc @gabrielenava

gabrielenava, 2024-09-23:
hi @AntonioConsilvio! the fault button works now! you can close the issue, thank you!

closed 2024-10-22 by AntonioConsilvio