robotology/mesh-support #1904: [Check microphone availability]

CarmelaCalabrese, 2024-09-17:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

See below.

### Detailed context

Me and @steb6 have to run a demo in _Festival della Scienza_ and we would need the microphone to interact with ergocub. I know that robot's "ears" need to be connected. Could it be possible to wire these microphones?

Thank you 🌻

AntonioConsilvio, 2024-09-19:
Hi @CarmelaCalabrese @ste93, the speaker should be connected to the head, while the microphones (ears) should be connected to the torso.

Let me know if everything works as expected!

S-Dafarra, 2024-09-25:
Hi @AntonioConsilvio, we tried testing the microphone but it does not seem to be working. The microphone volume from ``alsamixer`` is at maximum, but even by tapping on the "ears", the input value stays at 2%

CarmelaCalabrese, 2024-10-10:
Dear @AntonioConsilvio , is there any way to solve this issue?

Gandoo, 2025-05-16:
ciao, Removing the nape cover the MIC cable that goes down to the COM_Express on the Torso appeared to be damaged due to crushing between the cover and chassis.
I shortened the cable and retested, below @filippoborgogni  ran the test and it seems to be working.

[image]

cc @AntonioConsilvio @S-Dafarra @CarmelaCalabrese

closed 2025-05-16 by filippoborgogni