robotology/mesh-support #1906: Support for upcoming demos

gabrielenava, 2024-09-17:
### Robot Name 🤖

iCubGenova04 (iRonCub1) S/N:031

### Request/Failure description

iRonCub-mk1 is involved in a demo the 24th September and will also be sent to IROS conference after that. 

It would be better to resolve the three issues still opened:

https://github.com/robotology/icub-tech-support/issues/1665
https://github.com/robotology/icub-tech-support/issues/1666
https://github.com/robotology/icub-tech-support/issues/1591

### Detailed context

Concerning the three issues:

- https://github.com/robotology/icub-tech-support/issues/1665 is not a real issue, but I remember there might be another battery charger somewhere. This will help because iRonCub1 will use the crane of iRonCub3 during the demos, that will remain without crane for more than a month.

- https://github.com/robotology/icub-tech-support/issues/1666 I don't think this issue has been investigated before, might be just a cable.

- https://github.com/robotology/icub-tech-support/issues/1591 this is the bigger issue and also the most problematic one, because it might require to replace the right ankle board. Also, the FT is required for walking demos.

### Additional context

cc @S-Dafarra @DanielePucci

S-Dafarra, 2024-09-18:
It would be also useful if at least one camera is working

gabrielenava, 2024-09-23:
issues https://github.com/robotology/icub-tech-support/issues/1666 and https://github.com/robotology/icub-tech-support/issues/1591 are solved

issue https://github.com/robotology/icub-tech-support/issues/1665 is not blocking for the demo

issue https://github.com/robotology/icub-tech-support/issues/1912 has been opened for the cameras

I close this issue, thank you!

closed 2024-09-23 by gabrielenava