robotology/mesh-support #1907: Cannot staart the robot

S-Dafarra, 2024-09-19:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

While doing classical teleoperation, the robot stopped suddenly while looking to the right. ``yarprobotinterface`` was not responsive anymore and I had to kill the motors.
After that, the robot was not starting anymore even after restarting the motors.

### Detailed context

Here the log 
[robotNotStarting_log.txt](https://github.com/user-attachments/files/17063438/robotNotStarting_log.txt)

The error is 
```
[WARNING] from BOARD 10.0.1.21 (head-eb21-j2_3) time=445s 90m 231u :  SYS: the TX phase of the control loop has last more than wanted. TX execution time 2461[usec]. Latest previous execution times[usec] (RX=N/A, DO=N/A, TX=2461); (Rx=131, DO=104, ...). Num of CAN frames[CAN1,CAN2]=[64 0]
[INFO] Closing device head-eb21-j2_3-mc
[ERROR] from BOARD 10.0.1.21 (head-eb21-j2_3) time=445s 93m 104u :  SYS: EOtheCANservice could not tx frames on CAN bus. CanPort=CAN1. Size of fifo is 64
```

### Additional context

cc @illkid3 @mebbaid @GiulioRomualdi

S-Dafarra, 2024-09-19:
Cc @maggia80 @DanielePucci

AntonioConsilvio, 2024-09-19:
Hi @S-Dafarra, The CAN RFE board should be connected to the eb21 board. 
There are two possibilities, either the CAN cable is damaged or there is a 12V power supply which should not be there and which causes an occasional malfunction.

If the robot is to be used and the RFE is not strictly necessary, in the meantime you could temporarily remove it from the configuration files so that the robot can start, and in the meantime we will plan the intervention.

S-Dafarra, 2024-09-19:
Thanks a lot @AntonioConsilvio. I wonder if this can also cause the malfunction of the realsense

AntonioConsilvio, 2024-09-30:
> Thanks a lot [@AntonioConsilvio](https://github.com/AntonioConsilvio). I wonder if this can also cause the malfunction of the realsense

Hi @S-Dafarra, I would exclude a correlation because although they are both in the head, they have a different power and communication system. 

Me and @filippoborgogni removed the 12V connection on the RFE. Now it seems the RFE is working properly and should no longer cause this problem. Closing! ✅

closed 2024-09-30 by AntonioConsilvio