robotology/mesh-support #1908: Robot does not start

isorrentino, 2024-09-20:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The board 10.0.1.2 is not communicating. With FirwareUpdater I'm not able to see any board attached.

[image]

### Detailed context

[log_ergocub-torso_yarprobotinterface_24682.txt](https://github.com/user-attachments/files/17072649/log_ergocub-torso_yarprobotinterface_24682.txt)

S-Dafarra, 2024-09-20:
The issue seems on the left arm FT whose cable got damaged. The issue of no CAN boards visible from FirmwareUpdater seems due to the usual issue thaat it is not possible to detect CAN boards using FirmwareUpdater from the torso

S-Dafarra, 2024-09-20:
With https://github.com/icub-tech-iit/robots-configuration/commit/a182c7355a7ddbbf00525f8e7ff54e3f714c997e I have removed both arm FTs and the robot starts

S-Dafarra, 2024-09-20:
Cc @Gandoo

Gandoo, 2024-09-23:
ciao @AntonioConsilvio   I disconnected the left ft sensor at Can-FT and the right one from EB1. Need to consider replacing both FTSensors because the cable has come out of the housing and they might be stripped or torn, consequently the 5Vdc might give problems.

cc @isorrentino @DanielePucci

AntonioConsilvio, 2024-10-22:
Hi, there are already two issues (one in each arm) regarding the malfunctioning of FTs. I therefore proceed to close this issue in favour of:

- https://github.com/robotology/icub-tech-support/issues/1909
- https://github.com/robotology/icub-tech-support/issues/1923

closed 2024-10-22 by AntonioConsilvio