robotology/mesh-support #1910: Right elbow cable got broken

GiulioRomualdi, 2024-09-20:
### Robot Name 🤖

iCubGenova04 (iRonCub1) S/N:031

### Request/Failure description

While testing the walking controller on iCubGenova04 (iRonCub1) the right cable of the elbow got broken

### Detailed context

I think this is due to the high backlash in the shoulder, as the joint goes into resonance while walking, causing the cable to break.

### Additional context

[image]
[image]

### How does it affect you?

This will affect the important demo of next week 

@S-Dafarra @DanielePucci @maggia80 @Fabrizio69

Gandoo, 2024-09-23:
ciao @AntonioConsilvio On Friday the 20th I fixed the left elbow cable and checked the tendon tensions on the right shoulder but there seems to be problems on the mechanical

@Fabrizio69 @DanielePucci

AntonioConsilvio, 2024-09-25:
Hi! With the help of @Fabrizio69 and @Elio75 we replaced the broken tendon of right elbow. The elbows works as expected now! Closing ✅

closed 2024-09-25 by AntonioConsilvio