robotology/mesh-support #1911: Robotology-superbuild Upgrade

ddetommaso, 2024-09-23:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

We would need to upgrade the robotology-superbuild on the robot from: **v2023.05.2** to the latest (currently **v2024.09.0**)

### Detailed context

In order to run the stanford demo with the latest developments of the IK modules in ARE we need to align the robotology superbuild with the latest version.

cc @martinaxgloria @Nicogene @pattacini 

### Additional context

```
icub@icub-head:~$ lsb_release -a

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal
```

```
icub@icub-head:~$ uname -a

Linux icub-head 5.4.0-196-lowlatency #216-Ubuntu SMP PREEMPT Thu Aug 29 14:12:17 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```
icub@icub-head:/usr/local/src/robot/robotology-superbuild$ git status

HEAD detached at v2023.05.2
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	oldInstall/

nothing added to commit but untracked files present (use "git add" to track)
```

### How does it affect you?

This is pretty urgent because the robot is going to be shipped in the next few days.

AntonioConsilvio, 2024-09-23:
Hi, in order to plan any technical support we need to know the shipping date of the robot (if there is an official date).

At the moment we do not have the possibility to plan the support, so I suggest you try to update the robotology-superbuild according to this guide and give us feedback:

- https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#update

ddetommaso, 2024-09-23:
Hi @AntonioConsilvio,

before updating the robot I would like to ask you if it's just a software update on the head or additional steps are needed due to the upgrade considering the Ubuntu version and the robotology-build currently in use.

For the shipping date we should have soon an official date from @maggia80

AntonioConsilvio, 2024-09-24:
Hello @ddetommaso! The robotology-superbuild does not require any further software upgrades.

The only compatibility to check is with the robot's electronic boards. If you upgrade the robotology-superbuild, you must also update the firmware of the electronic boards, see repo version:

- https://github.com/robotology/icub-firmware-build/tree/v1.40.0

The update process is quite long and manual (to be done manually for each board). If you want, you can use a script that automatically updates all the boards (icub-firmware-build/scripts), but this will still take a few hours.

Another important thing: unfortunately we do not have any technicians available in the next few days to send to Erzelli in case of problems with the update and for this type of problems it is quite difficult to give remote support. 
For this reason, we advise you not to update a few days before the robot's shipping, unless it is absolutely necessary and essential for the demo. So if you think it is necessary to update the robot before the departure, probably you will have to do it by yourself.

AntonioConsilvio, 2024-12-18:
Hi! We have upgraded the robotology-superbuild and firmware at Stanford. The robot works properly and is able to do the demo.

Closing! ✅

closed 2024-12-18 by AntonioConsilvio