robotology/mesh-support #1913: Robot head keeps restarting

ddetommaso, 2024-09-23:
### Robot Name 🤖

iCubErzelli01 S/N:032

### Request/Failure description

We initially noticed two consecutive robot crash (robot falling down) in two different sessions.
After further investigation we noticed that the head is restarting and keeps restarting in some conditions. 

### Detailed context

We tested the status of the robot head running only yarpdev to see video streaming of the camera without launching the yarprobotinterface. In this condition the robot head after some minutes restarts. The issue seems get worse if also the motors are ON. 

### Additional context

[syslog.txt](https://github.com/user-attachments/files/17096358/syslog.txt)

[image]

### How does it affect you?

We must finalize a new experimental study with the robot within this week.

ddetommaso, 2024-10-01:
After further test we noticed that the issue is occurring quite systematically having the two yarpdev of the camera (right and left) running at the same time on the icub-head.

At this point we would need to schedule your intervention possibly on the 7th of October.

AntonioConsilvio, 2024-10-02:
Hi @ddetommaso, unfortunately, we will need October 7th to carry out a status check and pack the robot along with all the necessary materials for the trip to Stanford.

As a result, we won’t be able to schedule the intervention on that day.

However, we will try to come to Erzelli on one of the other days next week to perform the intervention. Would there be someone available as contact?

AntonioConsilvio, 2024-10-09:
Hi @mmwozniak, we are available tomorrow for the intervention. We will come in the morning around 8:30.

We will already be autonomous to start the intervention. We will then need you to join us for the robot test.

mmwozniak, 2024-10-09:
Thank you for information. I will be available to test the robot from around 10:00. 

Additional information: the robot also occasionally has problems with the neck - when turning off the yarp robot interface its head goes to the back and right, and then these motors have problems when starting again.

ddetommaso, 2025-01-21:
Hi @AntonioConsilvio,

this issue is not blocking for us and needs further investigation, because it seems that we cannot stream from both cameras at the same time. Please let us know when you will be check this.

AntonioConsilvio, 2025-05-15:
Hi @ddetommaso! We have recently worked on this robot a couple of times for different issues, and this problem has not reoccurred. If it didn't occur again during your experiments, I would proceed with closing the issue!

AntonioConsilvio, 2025-07-16:
Hi @ddetommaso, during the intervention performed in this issue:

- https://github.com/robotology/icub-tech-support/issues/2121

We had to rewire the whole head (including the power supply), so let us know if the problem persists after the complete rewiring

ddetommaso, 2025-07-16:
Hi @AntonioConsilvio,

since the issue is not recurring anymore we can close it.