robotology/mesh-support #1914: Right thumb not working for broken cable

GiuliaScorzaAzzara, 2024-09-25:
### Robot Name 🤖

iCubGenova03 S/N:020

### Request/Failure description

After default calibration, the right thumb got stuck. When trying to move it, the cable broke.

### Detailed context

Right thumb oppose, distal and proximal dof stopped working because of broken cable. Need to replace the cable.

reafrancesco, 2024-12-19:
@maggia80 @AntonioConsilvio since the right thumb never worked for more than half an hour after your fixing we agreed that there might something damaging the cable. Could you please come to have a look in the next days. We need to start a new experimentation and we need a working iCub. Thanks

filippoborgogni, 2024-12-23:
Hi @reafrancesco, I'm planning to come to have a look at the situation for this issue and for this one https://github.com/robotology/icub-tech-support/issues/1925
on 30 december !

@Fabrizio69 @AntonioConsilvio

filippoborgogni, 2025-01-07:
Hi @reafrancesco, I'm planning to come to have a look at the situation for this issue and for this one https://github.com/robotology/icub-tech-support/issues/1925 tomorrow!

@Fabrizio69 @AntonioConsilvio

reafrancesco, 2025-01-07:
Thank you so much. I confirm I will be in presence so to support your fixing if needed.
Thanks.

reafrancesco, 2025-01-21:
Hi @Fabrizio69 @AntonioConsilvio, when can we reschedule the repairs of the iCub since the previous attempt was not successful? It is important for us in order to plan activities that are now hindered by the failure.

filippoborgogni, 2025-01-30:
Hi @reafrancesco @GiuliaScorzaAzzara  , yesterday @edersonferrante and I replaced the right thumb oppose tendon. We tested it mechanically and it worked properly, but we were not able to test it with the `yarprobotinterface` , so let us know if the robot works when you will be able to test it !

AntonioConsilvio, 2025-02-18:
Hi @reafrancesco @GiuliaScorzaAzzara, @filippoborgogni and I came back to Erzelli last Wednesday as the robot was not starting properly.
We replaced an MC4 (2B2) board as the CAN channel had stopped working. In fact, the board was not found during calibration and the `yarprobotinterface` was crashing.

After finding out which firmware was correct for that specific board (see [guide](https://icub-tech-iit.github.io/documentation/icub_firmware/associated-firmware/associated-firmware/#cfw2can2-right-arm_2) for V2 arms), unfortunately the board was not of the correct revision (it was revision C). So we replaced it again (with revision D) and the robot started working properly again.

https://github.com/user-attachments/assets/9cbcdb4c-279e-4a35-bd3f-344394c09a5d

https://github.com/user-attachments/assets/d6091a03-a655-4ba8-8510-d095868f26d7

In order to get the thumb working properly, we had to make a few changes to the configuration files; I will do the PR shortly. The changes are already integrated into the robot head.

AntonioConsilvio, 2025-02-18:
Here the PR:

- https://github.com/robotology/robots-configuration/pull/712

Closing 🚀

closed 2025-02-18 by AntonioConsilvio