robotology/mesh-support #1915: LCD of the face is broken

steb6, 2024-09-25:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

the lcd (face) of the robot is completely black

### Detailed context

while using the robot, the screen started jiggling for a while and now it is completely black

steb6, 2024-10-14:
@AntonioConsilvio could this be fix before the end of the week? thanks

SimoneMic, 2024-10-21:
The LCD is still broken, and we need it at the Festival della Scienza.
Sorry for bothering you so much, thank you!

cc  @lornat75 @maggia80

filippoborgogni, 2024-10-22:
Hi, @SimoneMic, we replaced the flat cable that connects the display  to the MIP-1000 board because it was damaged.
For the moment the robot works!

SimoneMic, 2024-10-29:
During the demos the LCD behaved properly and everything went fine!

We can close this issue
Thank you!

closed 2024-10-29 by maggia80