robotology/mesh-support #1916: Right ring finger doesn't close well

martinaxgloria, 2024-09-26:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

During some grasping tests, I found out that the ring finger of the right hand doesn't close as expected.

### Detailed context

The ring finger closes before the pinky, so the grasp is not performed optimally.

### How does it affect you?

We are not able to perform the Stanford demo as well.

cc @Nicogene

fbiggi, 2024-10-01:
Tendini ripristinati 
Fingertip ripriatinatio e risaldati da @Gandoo 

[image]

[image]

@Fabrizio69 @AntonioConsilvio

martinaxgloria, 2024-10-02:
Until this morning, the issue about the ring/pinky seemed to be solved, but after a few calibrations of the hand, the problem of the tendon appeared again. 

cc @maggia80 @Nicogene @pattacini @Fabrizio69

Gandoo, 2024-10-02:
ho risaldato i fili sui 2 Fingertip che si erano staccati durante la sostituzione dei tendini.

[image]

cc @fbiggi @martinaxgloria  @Fabrizio69

Gandoo, 2024-10-02:
C'era un problema con l'abduzione del pollice Destro, si accostava tutto al palmo e non permetteva la riuscita della calibrazione. 
Ho verificato il cablaggio e movimento con la Jogger e alla fine mi si è staccato il filo bianco del sensore di Hall sul connettore Molex della scheda. Ricrimpato i fili, sostituito il Molex e riprovato a fare i movimento con jogger e funziona perfettamente.
In seguito @martinaxgloria ha verificato con @AntonioConsilvio i valori e in seguito la calibrazione ha cominciato ad andare a buon fine.

AntonioConsilvio, 2024-10-22:
During testing, the ring tendon broke. So I replaced the broken tendon.

The robot worked properly and the fingers closed at the same time. 
Now the robot has shipped to Stanford so I am proceeding with the closing of the issue. 🚀

closed 2024-10-22 by AntonioConsilvio