robotology/mesh-support #1918: ETH broken on right arm

S-Dafarra, 2024-09-27:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The eth backbone in the robot seems interrupted somewhere and the robot does not work well. The communication to the robot gets interrupted constantly

### Detailed context

@AntonioConsilvio  noted that the ETH cable in the right forearm seems broken

AntonioConsilvio, 2024-09-30:
Hi @S-Dafarra! The problem at the ETH was perhaps due to the damaged power cable of the MC4-PLUS c-shape EB22 board. Turning off temporarily could have caused the ETH communication to be interrupted. 

[image]

I fixed the cable, let's wait the next few days to see how the robot behaves.

AntonioConsilvio, 2024-10-11:
The problem has not reoccurred, closing! ✅

closed 2024-10-11 by AntonioConsilvio