robotology/mesh-support #1922: Neck yaw hardware fault

martinaxgloria, 2024-09-30:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

During the tests for the Stanford Demo, we noticed that the neck yaw went to hardware fault

### Detailed context

In certain positions of the head, the neck yaw went to HF MC overcurrent. I attached a figure of the cables that seem to be a little bit damaged.

[image]

### Additional context

To see if it's the cause of the problem, @Gandoo put some insulated tape and the problem seemed to disappear

### How does it affect you?

Before the shipping of the robot, maybe it might be necessary to fix the cable properly to not have the same issue once in Stanford

cc @Nicogene @pattacini @Fabrizio69

pattacini, 2024-09-30:
cc @ddetommaso

pattacini, 2024-09-30:
As agreed with @Gandoo, he will be replacing the flat cable when the other maintenance intervention takes place to repair the finger tendon, meaning tomorrow or the day after tomorrow.

Gandoo, 2024-10-01:
ciao @martinaxgloria @AntonioConsilvio   ho messo la guaina termorestringente sui cavi flat motori del collo M0 e M1 e del nastro isolamte a protezione del filo rosso 48Vdc del waist.

[image]

[image]

cc @pattacini @Nicogene @ddetommaso @Fabrizio69

pattacini, 2024-10-01:
Super! Thanks!

closed 2024-10-22 by AntonioConsilvio