robotology/mesh-support #1924: [Robot doesn't start because of problems on right leg]

fils99, 2024-10-02:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The robot doesn't start. From the log of the errors (see below), it seems the problem is related to right leg

### Detailed context

I was moving the right leg motors by using the motor gui, when at some point the motors went on fault. When I tried to restart the robot, I obtained the following error log.

### Additional context

[log_error.txt](https://github.com/user-attachments/files/17232391/log_error.txt)

Gandoo, 2024-10-03:
Smontando la cover anteriore coscia DX si vede il connettore CAN sulla EB6 staccato.

[image]

cc @Fabrizio69 @maggia80 @DanielePucci @isorrentino @fils99

Gandoo, 2024-10-03:
oltre al connettore staccato cè anche  un componente evidenziato con la freccia

 [image]

cc @Fabrizio69 @maggia80 @DanielePucci @isorrentino @fils99

Gandoo, 2024-10-03:
Sono riuscito a recuperare la scheda EMS4 risaldare connettore Molex e componente.
Come si vede la scheda ha subito un trauma il componente nel riquadro rosso è leggermente inclinato a differenza di quello nel riquadro giallo.

[image]

cc @Fabrizio69 @maggia80 @DanielePucci @isorrentino @fils99

Gandoo, 2024-10-03:
Nel frattempo, @AntonioConsilvio c'era anche il filo rosso del sensore FT della coscia staccato sul connettore, ho ricrimpato e sostituito il Molex da 6

Gandoo, 2024-10-03:
con @isorrentino abbiamo acceso e provato a fare la calibrazione e risultato è positivo. Ho ricoverizzato la gamba.

cc @Fabrizio69 @maggia80 @DanielePucci  @fils99

AntonioConsilvio, 2024-10-22:
Good job! Closing! 🚀

closed 2024-10-22 by AntonioConsilvio