robotology/mesh-support #1925: Broken right knee and broken right thumb

lauratriglia, 2024-10-04:
### Robot Name 🤖

iCubGenova07 S/N:040

### Request/Failure description

The right knee joint is broken, we give command to reach a position and the position is not reached and it becomes unstable. 
The right thumb has the cable broken. 

### Detailed context

Right knee might be related to the electronics, controlling the right leg. The right thumb instead is the cable that it's broken.

AntonioConsilvio, 2024-10-09:
Hi @lauratriglia tomorrow we will be in erzelli. We will try to provide support for this issue after trying to solve this one:

- https://github.com/robotology/icub-tech-support/issues/1913

If we can work on your robot, we will first start with the thumb and then move on to the knee.

@reafrancesco

AntonioConsilvio, 2024-11-29:
Hi! We intervened on the robot today, there were a few snags, but the problems on the hand seem to be solved. However, we wanted to finish testing and make sure everything is working properly, we were planning to come back Monday morning!

MissingSignal, 2024-11-29:
Dear @AntonioConsilvio,
thank you for your great support !
See you on Monday morning then 😉

reafrancesco, 2024-12-19:
@maggia80 @AntonioConsilvio since the right thumb never worked for more than half an hour after your fixing we agreed that there might something damaging the cable. Could you please come to have a look in the next days. We need to start a new experimentation on this iCub too and we need this working iCub for the specific experiment. Thanks

filippoborgogni, 2024-12-23:
Hi @reafrancesco, I'm planning to come to have a look at the situation on 30 december !

@Fabrizio69 @AntonioConsilvio

filippoborgogni, 2025-01-07:
Hi @reafrancesco, I'm planning to come to have a look at the situation for this issue and for this one https://github.com/robotology/icub-tech-support/issues/1914 tomorrow!

@Fabrizio69 @AntonioConsilvio

reafrancesco, 2025-01-07:
Thank you so much. I confirm I will be in presence so to support your fixing if needed.
Thanks.

reafrancesco, 2025-01-21:
Hi @Fabrizio69 @AntonioConsilvio, when can we reschedule the repairs of the iCub since the previous attempt was not successful? It is important for us in order to plan activities that are now hindered by the failure.

AntonioConsilvio, 2025-01-22:
Hi @reafrancesco, we have worked on this issue several times, but unfortunately we have not been able to close the issue yet. We are sorry for the inconvenience.

We have planned to come every Wednesday to Erzelli for the next two months to guarantee the presence of one or more technicians each week. At the end of the two months we will evaluate whether to extend our presence in Erzelli.

We will try to handle all issues, you will receive updates when we work on yours.

AntonioConsilvio, 2025-02-18:
Hello! As the right lowerarm of this robot  keeps giving problems. We decided to take the lowerarm to work on it at CRIS and try to restore it as best we could.

We will try to carry out this activity as quickly as possible. We will keep you updated!

reafrancesco, 2025-02-24:
yes sure @AntonioConsilvio! We understand the situation. Just keep us posted with the latest so that we can organize our experiments.

AntonioConsilvio, 2025-02-25:
Hi @reafrancesco, we finished the mechanical and electronic maintenance. Tomorrow we will face the testing phase.
If the tests are passed successfully, the plan is to go to Erzelli on Thursday 27 February (in two days' time) to reassemble the lowerarm! We will keep you updated.

reafrancesco, 2025-02-26:
Thank you. We are available to support you here in the lab on the 27th if your procedure goes as expected.

AntonioConsilvio, 2025-02-26:
Unfortunately, the test phase was not successfully passed 😕. During testing, the screw holding the whole thumb broke:

https://github.com/user-attachments/assets/03e0af22-d65e-4b16-bfc8-bab65c03bf80

This will unfortunately extend the maintenance time, we will keep you updated.

AntonioConsilvio, 2025-03-04:
Hi @reafrancesco here is a summary of the situation.
 
## Mechanical maintenance:
 
- Replacement of the motor and the tendon of the proximal thumb joint. The motor had become unusable because the pulley had broken, so we had to replace the tendons. 
 
- Replacement of thumb oppose tendon (it was damaged)
 
- Replacement of distal thumb tendon with sheath replacement (both were damaged)
 
- Replacing the middle distal tendon (it was torn)
 
- Replacing the finger adduction tendon (it was damaged)
 
- Thumb screw replacement (mentioned in previous [comment](url)) with subsequent replacement of distal and proximal tendons.
 
- Ring and pinky tendon replacement (broken during test)
 
## Electronic maintenance:
 
- Resoldered numerous cables, skin, motors and encoders:

  [image]

  [image]
 
## Software maintenance:
 
- Several recalibrations (I'll do the PR shortly)
 
> [!important]
> - I finally understood why the tendons, motors and screw broke on the thumb.
I had noticed a strange movement of the thumb already at Erzelli, however only with the arm here did I get a chance to debug it.
As you can see from the video, the adduction and proximal joints of the thumb did not move well:
>
>    https://github.com/user-attachments/assets/a09795c3-c9ec-416e-bde8-fb7a1223e260
>
>    It seemed that the proximal did not follow the movement of the thumb oppose, rather it went in the opposite direction 
  preventing the correct movement.
> 
>    So I replaced the joints matrices (see [file](https://github.com/robotology/robots-configuration/blob/master/iCubGenova07/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml)) by copying them from iCubGenova11, from:
>
>
> <group name="COUPLINGS"> 
>                            
>       <param name ="matrixJ2M"> 
>          1.000   0.000   0.000   0.000
>          1.000  +1.000   0.000   0.000
>          0.000   0.000   1.000   0.000
>          0.000   0.000   0.000   1.000   
>       </param>
>
>       <param name ="matrixM2J"> 
>           1.000   0.000   0.000   0.000
>          -1.000  +1.000   0.000   0.000
>           0.000   0.000   1.000   0.000
>           0.000   0.000   0.000   1.000   
>       </param>
>
>   To
>
> <group name="COUPLINGS"> 
>                            
>       <param name ="matrixJ2M"> 
>          1.000   0.000   0.000   0.000
>         -1.000  +1.000   0.000   0.000
>          0.000   0.000   1.000   0.000
>          0.000   0.000   0.000   1.000   
>       </param>
>
>       <param name ="matrixM2J"> 
>           1.000   0.000   0.000   0.000
>          +1.000  +1.000   0.000   0.000
>           0.000   0.000   1.000   0.000
>           0.000   0.000   0.000   1.000   
>       </param>
>
>    It is not yet clear why it used to work and now it doesn't, we are investigating this internally, but this is the explanation for the    numerous and repeated interventions we have had to carry out in the last period on iCubGenova07.
>
>    What we certainly know is that the first matrices were defined as correct (and the second as incorrect) in this [comment](https://github.com/robotology/icub-tech-support/issues/1019#issuecomment-730148256).
  However, on iCubGenova07 the second matrices were present at the time of the robot's release (2019), only to be changed 
  about a year later in this [commit](https://github.com/robotology/robots-configuration/commit/514fa4fb3bdf168e50eb97154a2a229a1317cd5f).
>
>    However, during the testing of iCubWaterloo01 (after about a month) the second matrices were committed (see [PR](https://github.com/robotology/robots-configuration/commit/514fa4fb3bdf168e50eb97154a2a229a1317cd5f)), which are then the ones that ended up on iCubGenova11 and currently working on iCubGenova07.

All joints are working now!
We are now finishing testing Fingertips, skins and then we will put the covers back on the lowerarm.

AntonioConsilvio, 2025-03-05:
Hi @reafrancesco, the lowerarm is ready!

We finished with the test.

## Calibration:

https://github.com/user-attachments/assets/e58f86cb-9706-4084-8f19-adf43255693c

## Thumb test:

https://github.com/user-attachments/assets/0c3cb6fe-9f8a-49e1-bf12-7320cd306e99

## Skin:

##

> [!Important]
> Here the PR for the new configuration files:
>
> - https://github.com/robotology/robots-configuration/pull/718

We plan to be in Erzelli tomorrow to reassemble the lowerarm!

reafrancesco, 2025-03-05:
Thank you so much @AntonioConsilvio. I will be in Erzelli tomorrow to support you with it.

filippoborgogni, 2025-03-07:
Hi @reafrancesco @lauratriglia , yesterday @AntonioConsilvio and I went back to reassemble the right lowerarm on the robot. We noticed that the ring and pinky tendons of the left arm were damaged and we fixed them. 
As you can see in the video below, the calibration phase was successful!

https://github.com/user-attachments/assets/796cce75-ea85-40df-adc6-2ff791166fcf

https://github.com/user-attachments/assets/024f2d40-7a96-4a8d-bff5-0ca81bf811db

AntonioConsilvio, 2025-03-10:
Hi @reafrancesco, since this issue has become very long and with so many interventions on the hands I would kindly ask you if it would be possible to open a different issue for the knee problem. 
This would really help us to have a better tracking of problems and the amount of problems on each robots. Thanks in advance!

By the way, I have to extend the problem not only to the knee, but to the whole leg. In fact, other joints struggle to move, see for example hip roll:

https://github.com/user-attachments/assets/dc695fb3-72a0-46eb-8853-04eb72b42cad

It seems that they do not have enough current to be able to generate enough torque. We already tried to measure the tension on the 2FOCs, which is correct. We will try to investigate to find a solution.

As a reference for the new issue, you can use this comment without any problems.

cc @Fabrizio69

edersonferrante, 2025-05-16:
Hi @reafrancesco, me and @AntonioConsilvio after analyzing the problem we found the red wire of the damaged knee LCORE and resoldered it.
After that, the knee started working properly again.

https://github.com/user-attachments/assets/86b46f01-780e-4202-a22c-210649e740b1

https://github.com/user-attachments/assets/3e14c412-afee-47b1-9e7b-f4b4075aa73b

https://github.com/user-attachments/assets/e352b518-3335-4da0-9586-49e671c4de27

cc @Fabrizio69