robotology/mesh-support #1926: Mechanical issue with robot's neck

paliasgh, 2024-10-08:
### Robot Name 🤖

iCubWaterloo01 S/N:044

### Request/Failure description

Hello, we are having an issue with the neck of our robot. When the head moves to certain positions, pitch and roll motors get disabled as the roll DOF gets stuck (is difficult to move physically).

### Detailed context

When the problem happens, we can hear a clinking sound from pitch and roll motors (not sure if both or only one). Joint 0 (pitch) gives this "i2t limit exceeded" error and goes to HWF fault mode. Joint 1 goes to idle (second picture, sorry the photo is blurred).

[image]
[image]

When this happens Joint 1 is difficult to move at a certain point (near 0). The mechanism in the neck gets mechanically stuck somehow. You can see the issue and hear a noise in these videos. We took the belt out to see if it was the motor or the other side. It was the other side (the neck). The movement is not smooth, is noisy, and it may fully get stuck, requiring us to move it back and again forward to pass that point.

https://github.com/user-attachments/assets/c6fac6f5-43d6-4e8f-a8c7-cba9e60e5c4d
https://github.com/user-attachments/assets/b141bd96-70dc-4340-82ef-b17d015b69d6
https://github.com/user-attachments/assets/9cdbcee1-ec48-4fb8-a5dc-93786e779358

Thanks for your assistance!

paliasgh, 2024-10-23:
Hello @AntonioConsilvio, I just wanted to follow up on this issue and see if there is any update. Thank you very much!

paliasgh, 2024-11-11:
Hello @AntonioConsilvio  and @pattacini, sorry for bothering you again, we are still looking forward to hearing from you about this issue. This has significantly limited our robot's functionality. We appreciate your help.

AntonioConsilvio, 2024-11-13:
Hi @paliasgh sorry for the late reply.
Considering the several problems presented by your robot, we were thinking that the best solution might be to either come to you to work on the robot directly or have the robot come back to us so that we can fix it.

We will provide you with more information on this in the near future, I mean to ask you if you would agree to intervene with one of these two options. Thank you!

AntonioConsilvio, 2025-05-07:
Hi, sorry for the delay in replying, but so many urgent activities have taken up a lot of our time.
 
Me and @aazocar7 came to Waterloo from 2 to 9 February. We closed all three issues.
 
## Issue [Neck](https://github.com/robotology/icub-tech-support/issues/1926)
 
The issue was a damaged bearing which we successfully replaced. 
Now the neck does not lock:
 
https://github.com/user-attachments/assets/fdecc46f-782a-4fc4-957b-71cc727f3180

## Issue [Shoulder](https://github.com/robotology/icub-tech-support/issues/1822)
 
After some testing, we realised that we had to replace the AEA of the shoulder yaw.
It was challenging because we had to detach the arm from the robot in order to replace it, but we succeeded and the shoulder works optimally now!

## Issue [fingers](https://github.com/robotology/icub-tech-support/issues/1231)
 
We replaced tendons (distal left thumb and distal right index finger) and also a motor (distal right index finger)

https://github.com/user-attachments/assets/c1dd8d1e-a81c-4a78-abe0-78b49695c9d9

https://github.com/user-attachments/assets/48b46280-acc3-4bca-862a-b90948152d8b

## Extra

We rewired several damaged electrical connections, and fixed an hardware fault on eyelids.

https://github.com/user-attachments/assets/f71870e2-e117-4a42-a028-3a61761075cb

https://github.com/user-attachments/assets/59a0391a-a7bb-4f9c-9eeb-fd2125f2b12e
 
So we managed to solve all the issues, thanks for your essential support! Closing! ✅

closed 2025-05-07 by AntonioConsilvio