robotology/mesh-support #1927: left leg boards not found

LoreMoretti, 2024-10-09:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

When launching the `yarprobotinterface` the left leg boards are not found and therefore the robot does not start.

### Detailed context

The following board are not found:

- left_leg-eb8-j0_3 IP 10.0.1.8
- left_leg-eb9-j4_5 IP 10.0.1.9

### Additional context

log:

[log_ergocub-torso_yarprobotinterface_leftLegBoards.txt](https://github.com/user-attachments/files/17309786/log_ergocub-torso_yarprobotinterface_leftLegBoards.txt)

### How does it affect you?

Cannot start the robot

Gandoo, 2024-10-10:
ciao @LoreMoretti @AntonioConsilvio @isorrentino il problema era del cavo Eth leggermente fuori dal connettore della scheda EB8, ho sostituito il connettore e inserito bene nella sede.

[image]

cc @Fabrizio69 @DanielePucci

isorrentino, 2024-10-10:
I'm using the robot now and everything works fine. Thanks @Gandoo!

closed 2024-10-10 by Gandoo