robotology/mesh-support #1928: Robot doesn't start due to timeout from left leg board

SimoneMic, 2024-10-10:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

After launching the YRI with the usual configuration, the following error appears:
`[ERROR] theNVmanager::Impl::ask() had a timeout for BOARD left_leg-eb8-j0_3 IP 10.0.1.8 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion`
And it's printed continuously.

### Detailed context

The robot doesn't event start the calibration process.
When checking with the FirmwareUpdater all the boards are in application.
I've tried to restart the motors but to no avail.

fils99, 2024-10-10:
# Update - problem on three boards

As reported in the log below, we can see that problems are present on three boards:
```
[ERROR] theNVmanager::Impl::ask() had a timeout for BOARD left_leg-eb8-j0_3 IP 10.0.1.8 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion

...

[ERROR] theNVmanager::Impl::ask() had a timeout for BOARD left_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion

...

[ERROR] theNVmanager::Impl::ask() had a timeout for BOARD right_leg-eb7-j4_5 IP 10.0.1.7 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion
```

[log_problem_board.txt](https://github.com/user-attachments/files/17330506/log_problem_board.txt)

filippoborgogni, 2024-10-11:
Hi @SimoneMic, @fils99 ! We fixed the ETH cable E6 that connect the EMS EB6 to the EMS EB7. 
cc @AntonioConsilvio

closed 2024-10-11 by filippoborgogni