robotology/mesh-support #1932: Make wrists more stable

steb6, 2024-10-14:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Tighten up the wrists of the robot

### Detailed context

For the festival della scienza demo, the robot will need to hold cartoon boxes, for 22 demonstartion in three days.
Anyway, the wrists are very loose. We don't want to control them, we just need them to stay still, but at the moment they cant even hold a box.
It would be nice to let us know the procedure to tighten the wrists, so that we can do it autonomously

AntonioConsilvio, 2024-10-22:
Hi @steb6, unfortunately, it is not possible to achieve a good level of stiffness in the mechanism. 
The mechanism is a rather old revision, and in order to achieve the stiffness of the newer ergoCubs, an upgrade would have to be made. 

The upgrade has already been requested in this [issue](https://github.com/robotology/icub-tech-support/issues/1842), but no upgrade is currently planned.

steb6, 2024-10-23:
Thanks for the clarification @AntonioConsilvio !

closed 2024-10-23 by steb6