robotology/mesh-support #1933: Jetpack cable desoldered

gabrielenava, 2024-10-15:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

One of the jetpack electronics cable desoldered from the related board (it is the one were we put some tape in the picture below):

[image]

would it be possible to fix it? At the moment, the robot is downstairs in the AMI area.

### Detailed context

cc @Fabrizio69

Gandoo, 2024-10-17:
ho eseguito la riparazione del filo danneggiato risaldandolo sul pin del connettore.
ciao @gabrielenava poi ci farai sapere se funziona. Grazie

[image]

cc @filippoborgogni  @Fabrizio69  cc @AntonioConsilvio

gabrielenava, 2024-10-18:
Hello @Gandoo, I have tested it today and works, thank you!!

AntonioConsilvio, 2024-10-22:
Good job @Gandoo, Closing! 🚀

closed 2024-10-22 by AntonioConsilvio