robotology/mesh-support #1934: `l_knee` does not complete calibration

mebbaid, 2024-10-15:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

As per the title, we tried to run the robot several times but the `l_knee` kept stopping mid-calibration. We restarted the motors and the PC and the issue persisted.

### Detailed context

As per the title, we tried to run the robot several times but the `l_knee` kept stopping mid-calibration. We restarted the motors and the PC and the issue persisted.

In the end, for our experiment @isorrentino moved the knee manually while it was in idle and we were able to run the Walking this way (stationary and not taking any steps).

Detailed logs are attached below

isorrentino, 2024-10-16:
The joint does not move from the `yarpmotorgui`, the encoder reads well, but if I change the desired position or torque the motor does not move.

cc @Gandoo @Fabrizio69

Gandoo, 2024-10-16:
ciao @isorrentino @AntonioConsilvio 
verificato il problema ginocchio, ho trovato il filo bianco cavo TDB staccato alla 2 foc 8B1 ma il problema del ginocchio NON funzionante era il filo Rosso della fase del motore totalmente tranciata, come evidenziato in foto.

[image]

cc @mebbaid @DanielePucci @filippoborgogni

AntonioConsilvio, 2024-10-22:
Good job @Gandoo, Closing! ✅

closed 2024-10-22 by AntonioConsilvio