robotology/mesh-support #1935: [Right thumb broken]

ddetommaso, 2024-10-16:
### Robot Name 🤖

iCubGenova08 S/N:043

### Request/Failure description

During the robot calibration the right hand does not calibrate properly and it gets stuck.

For accessing the robot, clarification and additional information please refer to @SilviaMoretti and @Cater19

cc @AgnieszkaWykowska

### Detailed context

The issue appears without having noticed any critical problem in the previous sessions. It's important to mention that the problem seems recurring (see previous issues).

### Additional context

[log_icub-head_yarprobotinterface_1454.txt](https://github.com/user-attachments/files/17390692/log_icub-head_yarprobotinterface_1454.txt)

### How does it affect you?

Critical! We are currently testing in the rehabilitation center and the issue with the robot is blocking for us.

AntonioConsilvio, 2024-10-22:
Hi @ddetommaso, we understand the urgency, but unfortunately we do not have the availability to intervene this week.
However, we will try to intervene as soon as possible, we will update you next week.

AntonioConsilvio, 2024-10-23:
Hi @ddetommaso, by rescheduling all the activities for these three days, we have been able to get availability tomorrow morning from 8am to 12pm. 

We have to return to CRIS in the afternoon.

Are you available tomorrow morning at 8 am?

cc @AgnieszkaWykowska @Cater19 @SilviaMoretti

AgnieszkaWykowska, 2024-10-23:
Hi Antonio,

Thank you very much, this is very good news and very important for us.
I confirm that my team will be there in Boggiano Pico tomorrow at 8 AM.

Best wishes,
Agnieszka

--
Prof. Agnieszka Wykowska
Istituto Italiano di Tecnologia
Head of the Research Line “Social Cognition in Human-Robot Interaction”
Coordinator of Center for Human Technologies (CHT)
Address: via Enrico Melen 83, 16152 Genova, Italy
Phone: +39 010 8172242
Email: ***@***.******@***.***>
Website: https://www.iit.it/people/agnieszka-wykowska
Editor in Chief International Journal of Social Robotics<International%20Journal%20of%20Social%20Robotics>
TUM-IAS Hans-Fischer Senior Fellow<https://www.ias.tum.de/ias/wykowska-agnieszka/>

AntonioConsilvio, 2024-10-24:
Hi, me and @fbiggi repaired the broken tendon in the right distal thumb. We tested the robot and it is able to perform your demo. Closing! ✅

cc @AgnieszkaWykowska @ddetommaso @Cater19 @SilviaMoretti

closed 2024-10-24 by AntonioConsilvio

AgnieszkaWykowska, 2024-10-24:
This is great news! Thank you so much!
Agnieszka