robotology/mesh-support #1936: Robot left leg has play near the upper leg FT sensor

isorrentino, 2024-10-16:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

As per the title, the robot left leg has play near the upper leg FT sensor. @Gandoo discovered this while repeating a cable.

cc @Gandoo @fbiggi @fils99 

### Detailed context

https://github.com/user-attachments/assets/ea291b09-c747-4777-80d4-18fa60cc8469

https://github.com/user-attachments/assets/5b6e8efe-f852-483b-97fb-1c392025f2cc

fbiggi, 2024-10-17:
cc @edersonferrante @filippoborgogni

Gandoo, 2024-10-17:
ciao, abbiamo staccato l'elettronica, stavvato la gamba dal sensore FT, sostituito le 6 viti e messo il frenafiletti. Rimontato il tutto.

[image]

cc @isorrentino @DanielePucci @Fabrizio69

edersonferrante, 2024-10-17:
Sostituito viti da 4x12 (Cod. 5933) e bloccate con la loctite 243 ( frena filetti) sulla piastra Alias ECUB_003_P_024, Cod. 15799.
Intervento effetuato con successo e gioco eliminato.
@filippoborgogni 

[image]

AntonioConsilvio, 2024-10-22:
Good job guys 🚀 Closing! ✅

closed 2024-10-22 by AntonioConsilvio