robotology/mesh-support #1937: Interface segs fault when connection with `yarp.carrier.shmem` got closed

GiulioRomualdi, 2024-10-16:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

I opened the issue here just to keep track of it.
During the iros2024 demo we closed the walking controller that uses shmem for the remote control board connection and the wbd external contact force port connection.

This is the error

```
[DEBUG] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=19746s 971m 986u :  src LOCAL, adr 3,(code 0x04000001, par16 0x0001 par64 0x0000000000000001) -> DEBUG: tag01  SET CONTROLMODE
[DEBUG] from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=19746s 953m 221u :  src LOCAL, adr 0,(code 0x04000001, par16 0x0001 par64 0x0000000000000001) -> DEBUG: tag01  SET CONTROLMODE
[DEBUG] from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=19746s 969m 188u :  src LOCAL, adr 1,(code 0x04000001, par16 0x0001 par64 0x0000000000000001) -> DEBUG: tag01  SET CONTROLMODE
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/head/rpc:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/head/rpc:o to /ergocub/head/rpc:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/head/command:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/head/command:o to /ergocub/head/command:i
[DEBUG] |yarp.carrier.shmem| INTERRUPT
[INFO] |yarp.os.impl.PortCoreOutputUnit|/ergocub/head/stateExt:o| Removing output from /ergocub/head/stateExt:o to /walking-coordinator/remoteControlBoard/ergocub/head/stateExt:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/torso/rpc:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/torso/rpc:o to /ergocub/torso/rpc:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/torso/command:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/torso/command:o to /ergocub/torso/command:i
[DEBUG] |yarp.carrier.shmem| INTERRUPT
[INFO] |yarp.os.impl.PortCoreOutputUnit|/ergocub/torso/stateExt:o| Removing output from /ergocub/torso/stateExt:o to /walking-coordinator/remoteControlBoard/ergocub/torso/stateExt:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/left_arm/rpc:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/left_arm/rpc:o to /ergocub/left_arm/rpc:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/left_arm/command:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/left_arm/command:o to /ergocub/left_arm/command:i
[DEBUG] |yarp.carrier.shmem| INTERRUPT
[INFO] |yarp.os.impl.PortCoreOutputUnit|/ergocub/left_arm/stateExt:o| Removing output from /ergocub/left_arm/stateExt:o to /walking-coordinator/remoteControlBoard/ergocub/left_arm/stateExt:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/right_arm/rpc:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/right_arm/rpc:o to /ergocub/right_arm/rpc:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/right_arm/command:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/right_arm/command:o to /ergocub/right_arm/command:i
[DEBUG] |yarp.carrier.shmem| INTERRUPT
[INFO] |yarp.os.impl.PortCoreOutputUnit|/ergocub/right_arm/stateExt:o| Removing output from /ergocub/right_arm/stateExt:o to /walking-coordinator/remoteControlBoard/ergocub/right_arm/stateExt:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/left_leg/rpc:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/left_leg/rpc:o to /ergocub/left_leg/rpc:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/left_leg/command:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/left_leg/command:o to /ergocub/left_leg/command:i
[DEBUG] |yarp.carrier.shmem| INTERRUPT
[INFO] |yarp.os.impl.PortCoreOutputUnit|/ergocub/left_leg/stateExt:o| Removing output from /ergocub/left_leg/stateExt:o to /walking-coordinator/remoteControlBoard/ergocub/left_leg/stateExt:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/right_leg/rpc:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/right_leg/rpc:o to /ergocub/right_leg/rpc:i
[INFO] |yarp.os.impl.PortCoreInputUnit|/ergocub/right_leg/command:i| Removing input from /walking-coordinator/remoteControlBoard/ergocub/right_leg/command:o to /ergocub/right_leg/command:i
[DEBUG] |yarp.carrier.shmem| INTERRUPT
[ERROR] yarprobotinterface intercepted a segmentation fault caused by a faulty plugin:
[ERROR] /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_shmem.so(+0x791d) [0x7fe19555b91d]
Trace requested at /usr/local/src/robot/robotology-superbuild/src/YARP/src/yarprobotinterface/Module.cpp:77 by code called from:
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(_Z16yarp_print_traceP8_IO_FILEPKcj+0x3b) [0x7fe19e14311b]

```

### Detailed context

cc @traversaro @S-Dafarra

traversaro, 2024-10-17:
I was able to reproduce the error with this git diff:

~~~
diff --git a/src/devices/controlBoardRemapper/tests/controlBoardRemapper_t1_test.cpp b/src/devices/controlBoardRemapper/tests/controlBoardRemapper_t1_test.cpp
index 496474d2b..de296aa59 100644
--- a/src/devices/controlBoardRemapper/tests/controlBoardRemapper_t1_test.cpp
+++ b/src/devices/controlBoardRemapper/tests/controlBoardRemapper_t1_test.cpp
@@ -15,6 +15,8 @@
 #include <catch2/catch_amalgamated.hpp>
 #include <harness.h>
 
+#define YARP_PORT_TEST_CARRIER_UNDER_TEST "shmem"
+
 using namespace yarp::os;
 using namespace yarp::dev;
 
@@ -379,6 +381,7 @@ TEST_CASE("dev::ControlBoardRemapperTest", "[yarp::dev]")
         remoteAxesList.addString("axisC3");
         remoteAxesList.addString("axisA2");
 
+
         Bottle remoteControlBoards;
         Bottle & remoteControlBoardsList = remoteControlBoards.addList();
         remoteControlBoardsList.addString("/testRemapperRobot/a");
@@ -390,6 +393,7 @@ TEST_CASE("dev::ControlBoardRemapperTest", "[yarp::dev]")
 
         Property & opts = pRemoteRemapper.addGroup("REMOTE_CONTROLBOARD_OPTIONS");
         opts.put("writeStrict","on");
+        opts.put("carrier", YARP_PORT_TEST_CARRIER_UNDER_TEST);
 
         REQUIRE(ddRemoteRemapper.open(pRemoteRemapper)); // remotecontrolboardremapper open reported successful, testing it
 
@@ -565,6 +569,7 @@ TEST_CASE("dev::ControlBoardRemapperTest", "[yarp::dev]")
 
         Property & opts = pRemoteRemapper.addGroup("REMOTE_CONTROLBOARD_OPTIONS");
         opts.put("writeStrict","on");
+        opts.put("carrier",YARP_PORT_TEST_CARRIER_UNDER_TEST);
 
         REQUIRE(ddRemoteRemapper.open(pRemoteRemapper)); // remotecontrolboardremapper open reported successful, testing it
 
diff --git a/src/devices/networkWrappers/RemoteControlBoard/tests/RemoteControlBoard_test.cpp b/src/devices/networkWrappers/RemoteControlBoard/tests/RemoteControlBoard_test.cpp
index ccf29005c..7c542432a 100644
--- a/src/devices/networkWrappers/RemoteControlBoard/tests/RemoteControlBoard_test.cpp
+++ b/src/devices/networkWrappers/RemoteControlBoard/tests/RemoteControlBoard_test.cpp
@@ -42,6 +42,9 @@
 using namespace yarp::dev;
 using namespace yarp::os;
 
+#define YARP_PORT_TEST_CARRIER_UNDER_TEST "shmem"
+
+
 TEST_CASE("dev::RemoteControlBoardTest", "[yarp::dev]")
 {
     YARP_REQUIRE_PLUGIN("fakeMotionControl", "device");
@@ -104,6 +107,7 @@ TEST_CASE("dev::RemoteControlBoardTest", "[yarp::dev]")
             p_cfg.put("device", "remote_controlboard");
             p_cfg.put("local", "/local_controlboard");
             p_cfg.put("remote", "/controlboardserver");
+            p_cfg.put("carrier", YARP_PORT_TEST_CARRIER_UNDER_TEST);
             REQUIRE(ddnwc.open(p_cfg));
         }
 
@@ -196,6 +200,7 @@ TEST_CASE("dev::RemoteControlBoardTest", "[yarp::dev]")
             p_cfg.put("device", "remote_controlboard");
             p_cfg.put("local", "/local_controlboard");
             p_cfg.put("remote", "/controlboardserver");
+            p_cfg.put("carrier", YARP_PORT_TEST_CARRIER_UNDER_TEST);
             REQUIRE(ddnwc.open(p_cfg));
         }
 
[... 548 lines cut ...]
==70843==
==70843==
==70843== 6 errors in context 2 of 2:
==70843== Thread 32:
==70843== Syscall param write(buf) points to uninitialised byte(s)
==70843==    at 0x505F5AD: __libc_write (write.c:26)
==70843==    by 0x505F5AD: write (write.c:24)
==70843==    by 0x52AF3AD: write (OS_NS_unistd.inl:1158)
==70843==    by 0x52AF3AD: send_i (ACE.inl:222)
==70843==    by 0x52AF3AD: ACE::send_n_i(int, void const*, unsigned long, unsigned long*) (ACE.cpp:1586)
==70843==    by 0x4E2CCAC: ShmemHybridStream::accept() (in /home/traversaro/yarp/buildconda/lib/yarp/yarp_shmem.so)
==70843==    by 0x4E2BC28: ShmemCarrier::becomeShmemVersionHybridStream(yarp::os::ConnectionState&, bool) (in /home/traversaro/yarp/buildconda/lib/yarp/yarp_shmem.so)
==70843==    by 0x4BBF2F5: yarp::os::impl::Protocol::respondToHeader() (in /home/traversaro/yarp/buildconda/lib/libYARP_os.so.3.9.0)
==70843==    by 0x4BBFCE0: non-virtual thunk to yarp::os::impl::Protocol::open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/traversaro/yarp/buildconda/lib/libYARP_os.so.3.9.0)
==70843==    by 0x4BB28E3: yarp::os::impl::PortCoreInputUnit::run() (in /home/traversaro/yarp/buildconda/lib/libYARP_os.so.3.9.0)
==70843==    by 0x4BD48D4: theExecutiveBranch(void*) (in /home/traversaro/yarp/buildconda/lib/libYARP_os.so.3.9.0)
==70843==    by 0x4D06B64: execute_native_thread_routine (thread.cc:104)
==70843==    by 0x4FDFA93: start_thread (pthread_create.c:447)
==70843==    by 0x506CA33: clone (clone.S:100)
==70843==  Address 0x17a1580c is on thread 32's stack
==70843==  in frame #2, created by ShmemHybridStream::accept() (???:)
==70843==
==70843== ERROR SUMMARY: 7 errors from 2 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
~~~

traversaro, 2024-10-17:
Better error:

~~~
[INFO] |yarp.os.impl.PortCoreInputUnit|/testRemapperRobot/b/command:i| Removing input from /test/remoteControlBoardRemapper/testRemapperRobot/b/command:o to /testRemapperRobot/b/command:i
[DEBUG] |yarp.carrier.shmem| INTERRUPT
/home/traversaro/yarp/src/devices/controlBoardRemapper/tests/controlBoardRemapper_t1_test.cpp:159: FAILED:
  {Unknown expression after the reported line}
due to a fatal error condition:
  SIGSEGV - Segmentation violation signal

===============================================================================
test cases:   1 |   0 passed | 1 failed
assertions: 103 | 102 passed | 1 failed

==72961==
==72961== Process terminating with default action of signal 11 (SIGSEGV)
==72961==    at 0x4FB9B1C: __pthread_kill_implementation (pthread_kill.c:44)
==72961==    by 0x4FB9B1C: __pthread_kill_internal (pthread_kill.c:78)
==72961==    by 0x4FB9B1C: pthread_kill@@GLIBC_2.34 (pthread_kill.c:89)
==72961==    by 0x4F6026D: raise (raise.c:26)
==72961==    by 0x4F6031F: ??? (in /usr/lib/x86_64-linux-gnu/libc.so.6)
==72961==    by 0x4FB3F42: __lll_lock_wait (lowlevellock.c:45)
==72961==    by 0x4FBB0F0: lll_mutex_lock_optimized (pthread_mutex_lock.c:48)
==72961==    by 0x4FBB0F0: pthread_mutex_lock@@GLIBC_2.2.5 (pthread_mutex_lock.c:93)
==72961==    by 0x52D3C35: ACE_OS::mutex_lock(pthread_mutex_t*) (OS_NS_Thread.cpp:2024)
==72961==    by 0x4E05DC8: acquire (Mutex.inl:102)
==72961==    by 0x4E05DC8: ShmemInputStreamImpl::read(char*, int) (ShmemInputStream.cpp:158)
==72961==    by 0x4E05FEA: ShmemInputStreamImpl::read(yarp::os::Bytes&) (ShmemInputStream.cpp:208)
==72961==    by 0x4E03F2B: non-virtual thunk to ShmemHybridStream::read(yarp::os::Bytes&) (ShmemHybridStream.h:40)
==72961==    by 0x4AC0162: read (InputStream.cpp:34)
==72961==    by 0x4AC0162: yarp::os::InputStream::readFull(yarp::os::Bytes&) (InputStream.cpp:103)
==72961==    by 0x4AB0990: yarp::os::AbstractCarrier::defaultExpectAck(yarp::os::ConnectionState&) (AbstractCarrier.cpp:249)
==72961==    by 0x4B9E20B: yarp::os::impl::Protocol::write(yarp::os::SizedWriter&) (Protocol.cpp:314)
==72961==
==72961== HEAP SUMMARY:
==72961==     in use at exit: 987,629 bytes in 8,855 blocks
==72961==   total heap usage: 677,605 allocs, 668,750 frees, 68,262,482 bytes allocated
==72961==
==72961== LEAK SUMMARY:
==72961==    definitely lost: 0 bytes in 0 blocks
==72961==    indirectly lost: 0 bytes in 0 blocks
==72961==      possibly lost: 100,016 bytes in 116 blocks
==72961==    still reachable: 887,613 bytes in 8,739 blocks
==72961==                       of which reachable via heuristic:
==72961==                         length64           : 149,216 bytes in 164 blocks
==72961==                         newarray           : 4,344 bytes in 21 blocks
==72961==                         multipleinheritance: 11,504 bytes in 4 blocks
==72961==         suppressed: 0 bytes in 0 blocks
==72961== Rerun with --leak-check=full to see details of leaked memory
==72961==
==72961== Use --track-origins=yes to see where uninitialised values come from
==72961== ERROR SUMMARY: 6 errors from 1 contexts (suppressed: 0 from 0)
==72961==
==72961== 6 errors in context 1 of 1:
==72961== Syscall param write(buf) points to uninitialised byte(s)
==72961==    at 0x50375AD: __libc_write (write.c:26)
==72961==    by 0x50375AD: write (write.c:24)
==72961==    by 0x52873AD: write (OS_NS_unistd.inl:1158)
==72961==    by 0x52873AD: send_i (ACE.inl:222)
==72961==    by 0x52873AD: ACE::send_n_i(int, void const*, unsigned long, unsigned long*) (ACE.cpp:1586)
==72961==    by 0x4E049B0: send_n (ACE.inl:184)
==72961==    by 0x4E049B0: send_n (SOCK_Stream.inl:115)
==72961==    by 0x4E049B0: ShmemHybridStream::accept() (ShmemHybridStream.cpp:91)
==72961==    by 0x4E03C28: ShmemCarrier::becomeShmemVersionHybridStream(yarp::os::ConnectionState&, bool) (ShmemCarrier.cpp:79)
==72961==    by 0x4B9E45F: yarp::os::impl::Protocol::respondToHeader() (Protocol.cpp:510)
==72961==    by 0x4B9EB18: yarp::os::impl::Protocol::open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (Protocol.cpp:220)
==72961==    by 0x4B949C3: yarp::os::impl::PortCoreInputUnit::run() (PortCoreInputUnit.cpp:99)
==72961==    by 0x4BADE64: theExecutiveBranch(void*) (ThreadImpl.cpp:81)
==72961==    by 0x4CDEB64: execute_native_thread_routine (thread.cc:104)
==72961==    by 0x4FB7A93: start_thread (pthread_create.c:447)
==72961==    by 0x5044A33: clone (clone.S:100)
==72961==  Address 0x179ed82c is on thread 32's stack
==72961==  in frame #2, created by ShmemHybridStream::accept() (ACE.inl:50)
==72961==
==72961== ERROR SUMMARY: 6 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
~~~

traversaro, 2024-10-18:
I provided a fix (or workaround, not sure) at https://github.com/traversaro/yarp/tree/fixshmem, if this happens again feel free to cherry-pick the fix https://github.com/traversaro/yarp/commit/bff26063438c93079acfd8dd929de14000b5fb58 on top of the yarp branch used on the robot. @S-Dafarra @GiulioRomualdi

traversaro, 2024-10-25:
I opened https://github.com/robotology/yarp/pull/3147 to fix the problem with `shmem` upstream in YARP. I also found and fixed the problem with `unix_stream` and `multipleanalogsensorclient`, so if we have problems with `shmem` we can also test again `unix_stream`, if you want to cherry-pick the changes the related commits are:
* https://github.com/robotology/yarp/pull/3147/commits/74b631fab6391e61233bca011cdaa026ad2bdb4a
* https://github.com/robotology/yarp/pull/3147/commits/37e1aab648224be63b4764b83ec17c4843e9614d

traversaro, 2025-01-21:
> This issue has been automatically marked as stale because it did not have recent activity. It will be closed if no further activity occurs.

The issue was fixed in https://github.com/RoboStack/ros-humble/issues/249 and released as part of YARP 3.10.0 .

closed 2025-01-21 by traversaro