robotology/mesh-support #1940: Left wrist pitch joint broken

SimoneMic, 2024-10-17:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

When starting the robot with the normal calibration file `ergocub.xml`, the motor of the left wrist spins loosely, without moving the joint, failing the calibration of the joint. Probably the belt is loose or badly connected (?)
I've noticed that the joint is more loose even when not powered on, or not controlled. See videos in detailed context.

In the meanwhile the calibration of the left wrist has been disabled: `<param name="CALIB_ORDER"> (3) (2) (0) (1) (7 8) (10 11 12) </param>`

 

### Detailed context

When not controlled, the wrists behave in this way (under the same external force)
Left wrist:
https://github.com/user-attachments/assets/7d4a1ce7-9701-412b-88bc-75b66e9f80ef
Right wrist:
https://github.com/user-attachments/assets/201031f7-3329-48c1-9924-906e59eeb078

### Additional context

Probably the issue #1932 was referring to the same thing, but wasn't clear/detailed enough.

### How does it affect you?

We need the wrists operable for the festival della scienza demo on the 23rd - 26th of October. Thank you so much and sorry for the inconvenience!
cc @lornat75 @maggia80

AntonioConsilvio, 2024-10-22:
Hi @SimoneMic, the left wrist has a broken belt and must not move or calibrate to avoid more damage. 
Replacing the strap takes too long to be done in time for the demo.

For the right wrist, instead, see comment:

- https://github.com/robotology/icub-tech-support/issues/1932#issuecomment-2429306424

steb6, 2024-10-30:
At Festival della Scienza, we fixed the left wrist of the robot with plastic ties to keep it more stable and improve the reliability of the box holding

AntonioConsilvio, 2024-12-17:
The broken belt was successfully replaced by @fbiggi, closing!

closed 2024-12-17 by AntonioConsilvio