robotology/mesh-support #1941: [Fake overheating motor l_knee]

fils99, 2024-10-18:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

When I run `yarprobotinterface`, the robot does not calibrate the `l_knee`.

### Detailed context

Below, the error:

```bash
[ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=154s 378m 40u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_knee (NIB=3), Raw_temperature_value=385)
```

Btw, using the motor GUI, I'm able to calibrate the `l_knee`, so everything seems fine.  Also touching the motor, it doesn't seem hot, so I think it's a fake overheating.

isorrentino, 2024-10-21:
[image]

Here we are again. But this time I cannot start the robot.

Gandoo, 2024-10-21:
ciao, scoverizzato la coscia SX e trovo il filo rosso staccato dalla TDB del ginocchio.
dopo aver risaldato il filo @isorrentino  fa il test e la calibrazione va a buon fine.
Rimontato le cover.

[image]

cc @fils99 @AntonioConsilvio @Fabrizio69

Fabrizio69, 2024-10-21:
Hi @fils99, this problem seems solved, I'm closing this issue... thanks @Gandoo

cc: @isorrentino @AntonioConsilvio