robotology/mesh-support #1942: The right-eye camera stopped working

mmwozniak, 2024-10-21:
### Robot Name 🤖

iCubErzelli01 S/N:032

### Request/Failure description

The right eye camera does not display image. The image is not completely black, but rather displays very dark video noise. The left eye camera does not show any problems.

 [image]

### Detailed context

On Thursday evening (17/10) the right-eye camera worked. On Friday morning after turning on the robot the camera did not work, but after changing the position of the robot's head when running the experimental script the camera started working again. However, after some time (maybe 10 minutes) it stopped working and hasn't regained the image since.

### Additional context

Log file:
[log_icub-head_yarprobotinterface_2338.txt](https://github.com/user-attachments/files/17458095/log_icub-head_yarprobotinterface_2338.txt)

### How does it affect you?

We use the right camera in our current experiment. We were able to switch to the left camera for the current purposes as a temporary solution, although the experimental setup was prepared to be used with the right camera

ddetommaso, 2025-01-21:
Hi @AntonioConsilvio,

this issue is not blocking for us but need to be fixed. Probably a flat cable of the camera needs to be replaced. We should have these spare parts in the lab if needed.

filippoborgogni, 2025-02-04:
Hi @ddetommaso , tomorrow @edersonferrante and I will come to work on the robot to try to solve this issue and this one https://github.com/robotology/icub-tech-support/issues/1976

filippoborgogni, 2025-02-06:
Hi @ddetommaso , yesterday @edersonferrante and I worked on this issue. We changed the flat cable of the right camera and in the photo below you can see the old one that it was really damaged. 

[image]

Unfortunately, we were not able to test it because we had some problems with the `yarp connect`, so let us know if the robot works properly when you test it. Thanks!

cc @AntonioConsilvio

filippoborgogni, 2025-02-07:
Today @edersonferrante and went back to finish to fix [this](https://github.com/robotology/icub-tech-support/issues/1976) issue.
When we tested the robot, with the help of @ddetommaso, we were also able to verify the correct function of the right camera (as you can see in the photo below). 

[image]

I would say that this issue can be closed. 
Thanks!🚀

cc @AntonioConsilvio

closed 2025-02-07 by filippoborgogni