robotology/mesh-support #1943: FT of left foot front not working

isorrentino, 2024-10-21:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The FT sensor mounted under the left foot front streams constant values.

[image]

### Detailed context

[image]

cc @mebbaid @SimoneMic

Gandoo, 2024-10-22:
ho aperto la scarpa SX per verificare ma al momento non ho trovato fili rotti e/o schiacciati

Gandoo, 2024-10-22:
dopo una verifica con @isorrentino e @AntonioConsilvio risultano i canali del sensore ID.14 saturi, procedo con la sostituzione.
...e siamo a 150!!

Gandoo, 2024-10-22:
ciao, disassembled the foot and sole, replaced the FT sensor FT sensor cod.12487.A6 SN133 instead of the SN097 cod.12487.A3 
@isorrentino Updated ID and calibrated, now it works

[image]

[image]

cc @AntonioConsilvio @Fabrizio69 @DanielePucci

Gandoo, 2024-10-22:
ticket material used  https://github.com/icub-tech-iit/tickets/issues/3742#issuecomment-2429058881

closed 2024-10-22 by AntonioConsilvio

Gandoo, 2024-10-23:
ciao @isorrentino and @AntonioConsilvio  I replaced the RJ45 connector, but the problem is on the FED card which has a broken network connector.
 You have to remove the back to replace the board

[image]

cc @S-Dafarra @Fabrizio69