robotology/mesh-support #1944: Connector of ethernet cable robot side broken

isorrentino, 2024-10-22:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

As per the title

cc @Gandoo @Fabrizio69 

### Detailed context

As per the title

AntonioConsilvio, 2024-10-22:
maybe related with: 

- https://github.com/robotology/icub-tech-support/issues/1921

Gandoo, 2024-10-23:
ciao @isorrentino and @AntonioConsilvio I replaced the RJ45 connector, but the problem is on the FED card which has a broken network connector.
You have to remove the back to replace the board

[image]

cc @S-Dafarra @Fabrizio69

Gandoo, 2024-10-25:
ciao @isorrentino and @AntonioConsilvio I managed to replace the FED board now the network connector remains inserted

[image]

cc @Fabrizio69

AntonioConsilvio, 2024-10-29:
Good job @Gandoo, closing! 🚀

closed 2024-10-29 by AntonioConsilvio