robotology/mesh-support #1946: Robot joints are miscalibrated

GiulioRomualdi, 2024-10-24:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

This morning I tried to run a classical walking experiment. The robot is not able to walk anymore since the joints of the legs seems completelly miscalibrated. 

### Detailed context

https://github.com/user-attachments/assets/53cc2714-84f4-47ab-b245-1523b7e8c4cd

### How does it affect you?

The robot does not walk anymore. The calibration is required to make the robot walk. 

cc @isorrentino @LoreMoretti @DanielePucci @S-Dafarra

Fabrizio69, 2024-11-06:
Daniele told me that the hardware limits (pins) are bent and this is the cause of the joints not being calibrated.
To avoid this problem, Ines changed the offsets for hip pitch and ankle pitch, but for these joint the hardware limits are these:

[image]

Another possibility is the knee:
[image]

GiulioRomualdi, 2024-11-06:
Recalibrating the entire robot might not be the optimal solution. In the past, we encountered a miscalibration caused by a broken pin (see https://github.com/robotology/icub-tech-support/issues/1838).

Since we don’t know which specific joint is miscalibrated, I would avoid recalibrating all of them. Instead, we might focus on identifying the problematic joint (though I'm not sure how to do this) and addressing that one.

Fabrizio69, 2024-11-11:
Hi @GiulioRomualdi , we have verified that all hardware limits are correctly fixed (and there aren't bent dowel pins)

cc: @AntonioConsilvio

AntonioConsilvio, 2024-12-17:
Hi @GiulioRomualdi! Were you able to recalibrate the robot properly?

GiulioRomualdi, 2024-12-17:
We calibrated them at humanoids

AntonioConsilvio, 2024-12-18:
Great, let's get on with closing the issue!

closed 2024-12-18 by AntonioConsilvio