robotology/mesh-support #1950: Right jetpack FT sensor not detected

gabrielenava, 2024-10-30:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We tried to start the robot but the startup failed. Checking the logger, it turns out that the right jetpack FT sensor is not detected.

### Detailed context

Here is the message of the `yaprlogger`:

[image]

here is the result of `Firmwareupdater`: the sensor is not detected:

[image]

and here the cable connecting the FT. We inspected the cable to understand if it was damaged, but it does not seem so. We would need support to analyse the situation and understand why the ft is failing.

[image]

gabrielenava, 2024-10-31:
hello, we found another cable for the FT and replaced it, and the FT works! so, it was the cable despite it does not look damaged at first sight.

Before closing the issue, can we ask for a new cable as replacement? we took one from another setup.

filippoborgogni, 2024-11-05:
Perfect. Closing!

closed 2024-11-05 by filippoborgogni