robotology/mesh-support #1952: Right thumb issues

ddetommaso, 2024-11-05:
### Robot Name 🤖

iCubGenova08 S/N:043

### Request/Failure description

During the robot calibration the right hand does not calibrate properly and it gets stuck.

For accessing the robot, clarification and additional information please refer to @SilviaMoretti and @Cater19

cc @AgnieszkaWykowska

### Detailed context

Today icub's right hand gave problems again so it was no longer possible to record. In particular, the problem again seems to concern the thumb and the index finger which are NOT CONFIGURED in the yarpmotorgui. We attach the log file and the screen of the joints of the right hand following the last calibration I had them do.

[image]

### Additional context

[image]
[log_icub-head_yarprobotinterface_1464.zip](https://github.com/user-attachments/files/17636312/log_icub-head_yarprobotinterface_1464.zip)

AntonioConsilvio, 2024-11-07:
Hi @ddetommaso, we planned to carry out the intervention on Monday morning, from 8 a.m.

Are any of you available on that day?

cc @@AgnieszkaWykowska @SilviaMoretti @Cater19

ddetommaso, 2024-11-07:
Hi @AntonioConsilvio,

thank you for the feedback. We'll check internally if this slot is suitable for the center and will give to you our availability as soon as possible.

cc @davideghiglino

ddetommaso, 2024-11-08:
@AntonioConsilvio Mon 11th at 8:00AM is ok, @Cater19 will be available to allow you to make the intervention.

AntonioConsilvio, 2024-12-17:
Hi, sorry for the delay in updating the issue, but we managed to resolve this issue on the date we agreed. Closing! 🚀

closed 2024-12-17 by AntonioConsilvio