robotology/mesh-support #1956: Cannot calibrate the robot

steb6, 2024-11-08:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Cannot calibrate the robot because of the following error:

[image]

### Detailed context

[log_ergocub-torso_yarprobotinterface_2155.txt](https://github.com/user-attachments/files/17680292/log_ergocub-torso_yarprobotinterface_2155.txt)

AntonioConsilvio, 2024-11-12:
Hi @steb6 @SimoneMic, I tested the robot and it works correctly, the problem did not occur to me. 
However, I am keeping the issue open to see if the problem will happen again.

steb6, 2024-11-12:
Thanks @AntonioConsilvio , actually it might happen when the joints are close to the limit. I will update this issue when I use the robot again

AntonioConsilvio, 2024-12-17:
Hi @steb6, do you know if this issue happened again? Thank you!

steb6, 2024-12-17:
Hi @AntonioConsilvio, not really, we can close for now, thank you!

closed 2024-12-17 by steb6