robotology/mesh-support #1958: Cannot start the robot

isorrentino, 2024-11-11:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Today I tried to use the robot and it was not starting because of this error

[image]

### Detailed context

cc @Miche19 @S-Dafarra

SimoneMic, 2024-11-11:
I think that is the same issue mentioned here: https://github.com/robotology/icub-tech-support/issues/1956

isorrentino, 2024-11-11:
@SimoneMic I don't think so, the error is related to different boards.

AntonioConsilvio, 2024-11-12:
Hello @isorrentino @SimoneMic the problem originates from the AMC EB31 which has a small fault. Sometimes it does not start in `eApplication`, but in `eUpdater`.

For now there are no plans to replace the board, as a walkaround you can do the following steps.

Start `FirmwareUdater -a` on the robot torso. Select the board and click on `Set Def Boot eApplication` (1) and `Force ETH Application` (2).

[image]

The robot will run correctly until the problem occurs again.

The problem is tracked internally, so I'm closing this issue and if we will work on upgrading the robot in the future, we will proceed with replacing the board.

closed 2024-11-12 by AntonioConsilvio