robotology/mesh-support #1961: Cabling issues on left arm and head

S-Dafarra, 2024-11-14:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

While teleoperating the robot (without the right arm), we started having issues reading the encoders. In particular, I noticed the following errors

## Error 1
```
[ERROR]   FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb21-j2_3 with IP 10.0.1.21  even after 0.00565791 seconds: CANNOT PROCEED ANY FURTHER
```

## Error 2
```
[ERROR] HostTransceiver::addSetROP__(): ERROR in eo_transceiver_OccasionalROP_Load() for BOARD w/ IP 0.0.1.8 after all attempts with id:  [I, D, 3, 2,  , =,  , 0, x, 0, 1, 0, 0, 0, 0, 1, 2,  , -, >,  , I, N, D,  , =,  , 0, ,,  , T, A, G,  , =,  , e, o, p, r, o, t, _, t, a, g, _, m, c, _, j, o, i, n, t, _, i, n, p, u, t, s, _, e, x, t, e, r, n, a, l, l, y, m, e, a, s, u, r, e, d, t, o, r, q, u, e, 
[ERROR] theNVmanager::Impl::set() fails t->addSetROP() to BOARD left_leg-eb8-j0_3 IP 10.0.1.8 for nv ID32 = 0x01000012 -> IND = 0, TAG = eoprot_tag_mc_joint_inputs_externallymeasuredtorque
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 9, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  5106 22068 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  35583 32581 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addROPask(): eo_transceiver_OccasionalROP_Load() for BOARD w/ IP [1, 0, ., 0, ., 1, ., 2, 1, 
[WARNING] HostTransceiver::addROPask(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  64124 32580 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 9, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  63612 32580 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  35583 32581 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addROPask(): eo_transceiver_OccasionalROP_Load() for BOARD w/ IP [1, 0, ., 0, ., 1, ., 2, 1, 
[WARNING] HostTransceiver::addROPask(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  64124 32580 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 9, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  63612 32580 656
[ERROR] HostTransceiver::addSetROP__(): ERROR in eo_transceiver_OccasionalROP_Load() for BOARD w/ IP 0.0.1.21 after all attempts with id:  [I, D, 3, 2,  , =,  , 0, x, 0, 1, 0, 0, 0, 1, 1, 4,  , -, >,  , I, N, D,  , =,  , 1, ,,  , T, A, G,  , =,  , e, o, p, r, o, t, _, t, a, g, _, m, c, _, j, o, i, n, t, _, c, m, m, n, d, s, _, s, e, t, p, o, i, n, t, 
[ERROR] theNVmanager::Impl::set() fails t->addSetROP() to BOARD head-eb21-j2_3 IP 10.0.1.21 for nv ID32 = 0x01000114 -> IND = 1, TAG = eoprot_tag_mc_joint_cmmnds_setpoint
[ERROR] while setting velocity mode
[ERROR] |yarp.devices.controlBoard_nws_yarp| Error while trying to command a velocity move on joint group
[ERROR] HostTransceiver::addROPask(): ERROR in eo_transceiver_OccasionalROP_Load() for BOARD w/ IP [1, 0, ., 0, ., 1, ., 2, 1, 
[ERROR] theNVmanager::Impl::ask() fails res->addROPask() to BOARD head-eb21-j2_3 IP 10.0.1.21 for nv ID32 = 0x0100010e -> IND = 1, TAG = eoprot_tag_mc_joint_status_core_modes_ismotiondone
[WARNING] EthResource::getRemoteValue() cannot have a reply from BOARD head-eb21-j2_3 with IP 10.0.1.21 at attempt # 1 w/ timeout of 0.2 seconds
[ERROR]   FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD head-eb21-j2_3 with IP 10.0.1.21  even after 0.00573182 seconds: CANNOT PROCEED ANY FURTHER
[ERROR] Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 1 ) for  BOARD head-eb21-j2_3 (IP 10.0.1.21) 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
```

## Error 3
```
[ERROR] from BOARD 10.0.1.25 (left_arm-eb25-j11_12) time=10049s 332m 930u :  MC: external fault button pressed.
[ERROR] from BOARD 10.0.1.25 (left_arm-eb25-j11_12) time=10049s 333m 47u :  MC: external fault button pressed.
```
⚠ The HF button was not pressed

## Error 4
```
[WARNING] from BOARD 10.0.1.9 (left_leg-eb9-j4_5) time=9770s 220m 596u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 627[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[ERROR] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 154m 40u :  SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. Number of lost items is  36.
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 151m 332u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 151m 448u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 151m 565u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 151m 681u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 151m 793u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 151m 906u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 25u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 140u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 255u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 366u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 476u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 588u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 699u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 810u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 152m 922u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 36u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 146u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 261u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 371u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 481u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 594u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 705u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 815u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=9770s 153m 928u : SYS: detected a runtime error
```

## Error 5
```
[WARNING] from BOARD 10.0.1.23 (left_arm-eb23-j7_10) time=10036s 58m 237u :  MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. (Joint=l_ring_pinky_oc (NIB=3), encoderPort=0)
```

### Detailed context

Here the full log: [log_robot_issue_left_arm.zip](https://github.com/user-attachments/files/17753604/log_robot_issue_left_arm.zip)

Note, these errors are quite intermittent, might be depending on the motion of the robot, and the robot starts nonetheless.

S-Dafarra, 2024-11-15:
cc @Fabrizio69 @maggia80

AntonioConsilvio, 2024-12-18:
Hi @S-Dafarra, the 1 and 2 errors was related due an hardware problem on ETH cable that connects the 2 MC4-PLUS on the head (cable name; E20).

The error 3 and 4 was related and due a fault cable disconnected from the board.

Closing! 🚀

closed 2024-12-18 by AntonioConsilvio