robotology/mesh-support #1962: Realsense doesn't start

steb6, 2024-11-15:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

.

### Detailed context

When I try to launch "yarprobotinterface --config realsense2.xml" on ergocub-head, I get the following errors:

[INFO] |yarp.device.realsense2| Sensor warm-up...
[INFO] |yarp.device.realsense2| m_pipeline.wait_for_frames() failed with error: ( Frame didn't arrive within 15000 )

steb6, 2024-12-02:
Hi @AntonioConsilvio , today me and @CarmelaCalabrese tried to start the camera, but we still ended up with the same error as above. It seems that disconnecting the USB cable or restarting the robot dosn't help. Thank you!

SimoneMic, 2025-01-20:
Today when starting the realsense device, it hangs forever waiting for frames, giving this error in loop:
`[ERROR] |yarp.device.realsense2| m_pipeline.wait_for_frames() failed with error: ( Frame didn't arrive within 15000 )`
Yarp on `ergocub-head` is at version 3.10, and yarp-device-realsense2 is updated to master.
After rebooting the head the issue persisted.

It seems to me the usual connection issue, where the flat cable is slightly disconnected and it prevents the proper image streaming.
@AntonioConsilvio

mebbaid, 2025-02-04:
Today I am having the same issue with the `realsense`. Just reporting here to keep track

AntonioConsilvio, 2025-03-18:
Closing due this comment:

- https://github.com/robotology/icub-tech-support/issues/2062#issuecomment-2732074078

closed 2025-03-18 by AntonioConsilvio