robotology/mesh-support #1963: When closing yarprobotinterface, MAS_ERROR occurs

S-Dafarra, 2024-11-15:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

During IROS, we noticed that when closing the ``yarprobotinterface``, the following errors were printed
```
[ERROR] |yarp.device.multipleanalogsensorsserver| Failure in reading data from sensor head_imu_0 of type ThreeAxisGyroscope with code MAS_ERROR, no data will be sent on the port
```

### Detailed context

It might be possibly due to a wrong detach order in the configuration files

### Additional context

cc @valegagge @MSECode as we noticed this during IROS

MSECode, 2024-11-20:
I'm not sure if this problem is totally related to the bug solved with this PR: https://github.com/robotology/icub-main/pull/963, where we used to have core dumping problems at yri closure due to the fact of not cleaning some resources in the correct manner. However, it might be worth a check.

AntonioConsilvio, 2025-03-31:
Hi @S-Dafarra, do you know if this problem is still present on the robot?

traversaro, 2025-04-01:
By reading the configuration file, I guess the problem is that the `head-imuFilter` device `detach` action has a `shutdown` level of `5`, see https://github.com/icub-tech-iit/robots-configuration/blob/8883a01d17fbb6da01372843ede67b066ce91ffd/ergoCubSN002/wrappers/inertials/head-imuFilter.xml#L19, while the `head-imuFilter_wrapper` device `detach` action has a `shutdown` level of `6`, see https://github.com/icub-tech-iit/robots-configuration/blob/8883a01d17fbb6da01372843ede67b066ce91ffd/ergoCubSN002/wrappers/inertials/head-imuFilter_wrapper.xml#L19 . 

As `head-imuFilter` is a device that attaches to some devices and exposes interfaces to which other wrapper attach, I guess it is in some way equivalent to a `remapper`, so I think it make sense to use the `attach`/`detach` levels documented in https://github.com/robotology/robots-configuration/issues/377 for remappers to fix this problem.

closed 2025-04-03 by S-Dafarra