robotology/mesh-support #1964: CAN errors on left arm

S-Dafarra, 2024-11-15:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

During IROS we noticed the following errors:
```
18.310346 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=4892s 313m 104u :  SYS: EOtheCANservice could not tx frames on CAN bus. CanPort=CAN2. Size of fifo is 64"
18.310474 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=4892s 316m 728u :  SYS: EOtheCANservice could not add a CAN frame in tx fifo. CanPort=CAN2 Frame.ID=514, Frame.Size=2 Frame.Data=0x107
18.310483 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=4892s 317m 27u :  SYS: EOtheCANservice could not add a CAN frame in tx fifo. CanPort=CAN2 Frame.ID=525, Frame.Size=2 Frame.Data=0x35
```

### Detailed context

The robot was still functional, but this might indicate an issue on the CAN termination. Unfortunately, we don't have the full log.

cc @valegagge @MSECode as we noticed this during IROS

valegagge, 2024-11-18:
As so far as I remember the problem was due to a missing termination on the CAN.
as you can see from the log lines the FW could not add any CAN message, therefore this means that the transmission queue is full, so it seems that the CAN BUS is not operative and the problem is not strictly related to the battery pack since on the same bus there is also the FT sensor. 

@MSECode / @AntonioConsilvio / @AntonioAzocar could you please tell me if I'm remembering well? 

thanks

AntonioConsilvio, 2024-12-18:
Hi! Looking at the errors, the CAN where the error is present is CAN2.

There is an issue where it is reported that the FT of the left shoulder disconnects:

- https://github.com/robotology/icub-tech-support/issues/1965

So I assume the two errors are related, so I'm closing this one in favour of the other one!

closed 2024-12-18 by AntonioConsilvio

valegagge, 2024-12-18:
In this issue, it seems that the eb2 could no longer transmit on CAN2 bus. Its TX queue is full.

In the issue of FT sensor, it seems that the FT stopped transmitting.