robotology/mesh-support #1967: Robot joints are miscalibrated

isorrentino, 2024-11-18:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

We experienced the same with ergoCubSN001 https://github.com/robotology/icub-tech-support/issues/1946.

I do not attach photos, but the same problem is explained in the other [issue](https://github.com/robotology/icub-tech-support/issues/1946).

For now, I manually aligned the joints and changed deltas in robots-configuration, but this is not the solution as the robot is still miscalibrated.

### Detailed context

We should understand why movements with high accelerations cause this, and maybe it is worth having a way to re-calibrate the joints easily or quickly. 

cc @DanielePucci @S-Dafarra @mebbaid

S-Dafarra, 2024-11-19:
One thing we noticed is that the distance between the feet was not 48.8mm as it should be. But probably this was a problem already present before.

AntonioConsilvio, 2025-01-22:
Hi! I know that this robot has been used in the last period. So I would to asks if this problem is still present on the robot or you has been able to calibrate it again. Thank you!

filippoborgogni, 2025-05-20:
Hi @isorrentino,
today with @AntonioConsilvio  I checked the calibration of the robot.
We had to change the `offset` for the `right ankle pitch` joint (in `CalibrationZero`, in the file `calibrators/right_leg-calib.xml` ) as it did not stay perfectly in 0.
The robot is now firmly on the ground with all the leg joints in 0 and the calibration seems perfect.

Closing!🚀