robotology/mesh-support #1968: [Provide a meaningful TITLE, but DON'T enter the robot name here]

mbarborinii, 2024-11-19:
### Robot Name 🤖

iCubGenova02 S/N:001

### Request/Failure description

Left hand not calibrating properly
Index finger always gets stuck and does not recover a relaxed position
All joints from wrist onward don't move when prompted from yarpmotorgui

### Detailed context

Left hand not calibrating properly
Index finger always gets stuck and does not recover a relaxed position
All joints from wrist onward don't move when prompted from yarpmotorgui

### How does it affect you?

That makes me sad :(

gaurvigoyal, 2024-12-09:
We would like to add the "urgent" label to this ticket.
@AntonioConsilvio 

PI: @ChiaraBartolozzi 
Research line: EDPR

filippoborgogni, 2024-12-17:
### Update left arm :
we replaced the tendons (middle distal and index proximal), we also restored the connections of the encoders (middle and index distals).

### Update right arm :
we had to fix the CAN ID of mc4 boards that were set wrong, and now the robot starts properly.

The last remaining problem is the replace of the ring and pinky tendons, but we are waiting the purchase because we have not spare parts of tendons 

Cc: @AntonioConsilvio

AntonioConsilvio, 2025-02-14:
> The last remaining problem is the replace of the ring and pinky tendons, but we are waiting the purchase because we have not spare parts of tendons

Hi! @edersonferrante replaced the broken tendon and it works!

Since the last problem mentioned in this issue has also been solved, I proceed to close the issue ✅

closed 2025-02-14 by AntonioConsilvio