robotology/mesh-support #1972: Ethernet connection is unreachable

gabrielenava, 2024-11-26:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

After coming back from the Humanoids conference, we tried to turn on the robot to verify its status. Unfortunately, the `icub-head` ethernet connection does not work.

### Detailed context

We tried to change the ethernet cable, change the switch to which the ethernet was connected, change the server, change the laptop. None of this worked. We concluded that the network board on the robot probably got damaged and needs to be checked.

HosameldinMohamed, 2024-11-27:
We opened the back cover and indeed the Ethernet connection was broken

Fabrizio69, 2024-11-29:
⚒️ WIP:
- Previous fault with this board (cod.10143.A): see [this link](https://github.com/robotology/icub-tech-support/issues/1903) -> Solution: replaced board
- This time P2 connector (cod.1394) was torn off the PCB. This connector now is available in the warehouse, we'll try to replace it

Fabrizio69, 2024-11-29:
## OLD CONNECTOR

[image]

[image]

[image]

## NEW CONNECTOR

[image]

Fabrizio69, 2024-11-29:
@gabrielenava the particular is on my desk, ready for delivery

gabrielenava, 2024-11-29:
thank you @Fabrizio69, we tested it and it works!

Fabrizio69, 2024-11-29:
OK @gabrielenava , we can close this issue

cc: @AntonioConsilvio

closed 2024-11-29 by Fabrizio69