robotology/mesh-support #1973: Broken board support on the robot back

gabrielenava, 2024-11-26:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

After coming back from Nancy, the robot got damaged on the back. More specifically, there is a board completely disconnected from the support and held to the robot only through the electric wires.

### Detailed context

Here you can see the board I was mentioning:

[image]

it was connected to the metal part below with 4 plastic screws. All of them are broken, two of them broke inside the metal part.

[image]
[image]

gabrielenava, 2024-12-17:
I have found two spacers and mounted them. The board is now at least not floating anymore, but the support remains very weak. It would be nice to fix it with a more permanent solution.

cc @HosameldinMohamed @Giulero

edersonferrante, 2024-12-20:
Hi @gabrielenava, I replaced the screws and spacers

[image]
[image]