robotology/mesh-support #1975: Several fingers do not open or do not close.

S-Dafarra, 2024-11-27:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

I have noticed that a few fingers on both hands do not move as expected.

### Detailed context

In particular:
- the left thumb OC does not move and the motor encoder value is very high
- the left middle OC closes but it does not open. If I pull it I feel something making a lot of friction
- the left pinky OC closes but it does not open
- the right middle OC does not move and the motor encoder value is very high

### Additional context

All the fingers were checked and fixed before departure and have been used only once during a demo. They seem to have broken on their own.

cc @maggia80 @Fabrizio69 @fiorisi

S-Dafarra, 2024-12-05:
cc @fbiggi @steb6

AntonioConsilvio, 2025-01-24:
Hi! @edersonferrante restored all tendons of all fingers. All fingers are working properly now.
Closing! 🚀

closed 2025-01-24 by AntonioConsilvio