robotology/mesh-support #1976: Right thumb tendon broken

ddetommaso, 2024-11-28:
### Robot Name 🤖

iCubErzelli01 S/N:032

### Request/Failure description

One of the tendon of the right thumb seems to be broken.

### Detailed context

While playing arm movements the thumb just broke.
[log_icub-head_yarprobotinterface_2133.txt](https://github.com/user-attachments/files/17948974/log_icub-head_yarprobotinterface_2133.txt)

### Additional context

[image]
[image]

### How does it affect you?

It is not causing severe impact on data collection for the current experiment, but it might create it for the following one.

ddetommaso, 2025-01-28:
Hi @AntonioConsilvio 

let me update this about the impact because, after two months we open the issue, it became quite urgent since we will need to start experiment in the next couple of weeks.

filippoborgogni, 2025-02-04:
Hi @ddetommaso , tomorrow @edersonferrante and I will come to work on the robot to try to solve this issue and this one https://github.com/robotology/icub-tech-support/issues/1942

ddetommaso, 2025-02-04:
Hi @filippoborgogni,

thank you for taking care of this. As previously mentioned to @AntonioConsilvio, tomorrow I will be out of the office to follow another activity, but he ensured me that you can fix the issue autonomously. Please, after the fix, ensure that the robot is fully functional.

Thank you

filippoborgogni, 2025-02-06:
Hi @ddetommaso , yesterday @edersonferrante and I replaced the left thumb proximal that was broken. We tried it mechanically, with the JOGGER board and with the `yarprobotinterface` we tested all the joints of the left thumb. 
In the video below you can see that it worked properly.

https://github.com/user-attachments/assets/f8bebe44-5324-4962-9fa5-6dd03111bb16

We wait your feedback to close this issue. Thanks!

cc @AntonioConsilvio

ddetommaso, 2025-02-06:
Hi @filippoborgogni 

unfortunately we cannot still close this issue because after the fixing it seems that now the left wrist is not calibrating.
It was enough to do a few rounds of complete calibration of the robot to notice this problem.

[log_icub-head_yarprobotinterface_1148.txt](https://github.com/user-attachments/files/18693622/log_icub-head_yarprobotinterface_1148.txt)

We would need the robot functioning for the school day happening next Tuesday, could you please come in the next following days to finalize this?

Thank you

cc @AntonioConsilvio @maggia80

filippoborgogni, 2025-02-07:
Today @edersonferrante and I went back to fix this issue. 
We found a couple of unsoldered cables from the AEA board, we reconnected the cables and we verified that the wrist is working correctly (as you can see in the video below). 
We also found a problem on the encoder (hall sensor) of the proximal index tendon: so we found some unsoldered cables on the MAIS board, we fixed it and we verified that the sensor was reading correctly (the video shows the correct reading).

https://github.com/user-attachments/assets/37700bca-f950-4de8-aa2e-f5f6eb374cfb

https://github.com/user-attachments/assets/0bc02adf-b510-490f-9bc7-d8a8816563ce

After that, with @ddetommaso , we tested the robot and it worked properly. 
Closing! 🚀

cc @AntonioConsilvio

closed 2025-02-07 by filippoborgogni

maggia80, 2025-02-08:
edited since it was left thumb