robotology/mesh-support #1978: Screen is not functional

mebbaid, 2024-12-02:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

I turned on the robot this morning and the screen is barely legible. 

### Detailed context

See attached picture

[image]

### How does it affect you?

We are unable to monitor the status of the battery and the robot

S-Dafarra, 2024-12-10:
Just to give some context, the screen broke on the SN002 robot, and we swapped the broken screen with the one of the 001 (this is also the reason why the screen is reversed).

We noticed that this issue happened when pressing the cover to insert the battery. Looking inside the cover we noticed that when pushed, it could have pressed on the corner of the screen, thus breaking the LCD. @AntonioConsilvio used the Dremel to remove some material from the edge close to the screen
[image]

cc @maggia80 @marcellosavoldi

marcellosavoldi, 2024-12-16:
@Elio75   , we could report this problem to Rezonico who is designing the new covers.

cc: @Fabrizio69 , @fiorisi

edersonferrante, 2024-12-19:
Hi @mebbaid  , screen replacement performed

closed 2024-12-19 by edersonferrante