robotology/mesh-support #1980: Right arm ft not found

isorrentino, 2024-12-02:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The ft of the right arm is not found and is not possible to start the robot.

Here the error

[image]

### Detailed context

Here firmwareupdater

[image]

isorrentino, 2024-12-03:
cc @Gandoo @fbiggi

Gandoo, 2024-12-03:
Ciao, @isorrentino  @AntonioConsilvio   the FT sensor has a cut cable, this problem has already been highlighted that the cable could be damaged.
I try to cut the cable where it is broken and solder it with a new cable, we don't have a FT sensor to replace

[image]

cc @Fabrizio69 @DanielePucci @maggia80

Gandoo, 2024-12-03:
ciao @isorrentino
 i cut the FT sensor wire and soldered the wires together with a new wire, connected and awaiting testing.

[image]

[image]

[image]

cc @Fabrizio69 @DanielePucci @maggia80 @AntonioConsilvio

isorrentino, 2024-12-03:
Thanks @Gandoo, I'm going to test it. I'll keep you updated here.

isorrentino, 2024-12-03:
The robot starts without errors. Thanks again.

isorrentino, 2024-12-04:
This is solved @AntonioConsilvio, I think the ticket can be closed.

closed 2024-12-04 by Fabrizio69