robotology/mesh-support #1981: Torso pitch motor burned

isorrentino, 2024-12-02:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

I turned on the robot and the pitch of the torso started to smell making a strange noise, as if it was frying. After this, the measured motor current started to be 11 Ampere without moving the joint.

cc @DanielePucci 

### Detailed context

I do not have anything to add here.

isorrentino, 2024-12-03:
cc @Gandoo @fbiggi

Gandoo, 2024-12-03:
ciao @isorrentino @AntonioConsilvio 
Last week they did some tests with the turbines, they added weight on the upper and as a result the engine may have suffered damage due to the position holding!!

cc @Fabrizio69 @DanielePucci @maggia80

Gandoo, 2024-12-03:
ciao @isorrentino @AntonioConsilvio 
and the locking pin has been placed, it is necessary to remember to disable the engine pending verification and possible replacement.
Grazie!!

cc @Fabrizio69 @DanielePucci @maggia80

Gandoo, 2024-12-05:
ciao @isorrentino @AntonioConsilvio we checked the continuity of the 3 phases of the motor and they are fine, then we tried the movement with the box and it is fine, then we reconnected and it seems to work.
**The problem noted is that if the robot has its arms forward it struggles to return to an upright position unlike when they are perpendicular to the body**

cc @Fabrizio69 @DanielePucci @maggia80 @filippoborgogni @edersonferrante

isorrentino, 2024-12-05:
We tried to use the robot with the torso_pitch enabled, but it continuously went to hardware fault. @fbiggi put back the pin again and changed the configuration files for removing it from calibration.

isorrentino, 2024-12-09:
This problem is related to https://github.com/robotology/icub-tech-support/issues/1990.

We have determined that the motor burnouts were not caused by the experiments but rather by the new 2FOC firmware version released in https://github.com/robotology/icub-firmware-build/releases/tag/v1.41.0

cc @Fabrizio69 @DanielePucci @maggia80 @ale-git @Gandoo @AntonioConsilvio @GiulioRomualdi @fbiggi

GiulioRomualdi, 2025-02-07:
I comment the issue just to remove the stale label 🏷

isorrentino, 2025-03-03:
Hi @AntonioConsilvio, @Gandoo, @fbiggi, and @Fabrizio69, back in time, we thought this motor was burned, but we later discovered that the smoking was caused by a bug in the firmware. The same situation occurred with the left hip roll, and changing the firmware solved the problem without any damage to the motor. For this reason, does it make sense to re-check this motor now that the firmware bug is solved? This way, we can confirm its current condition for sure.

cc @mebbaid @GiulioRomualdi @S-Dafarra

GiulioRomualdi, 2025-03-14:
Hi all, Could it be possible to double-check the phase of the motors? If it is working as expected, we can remove the pin for the demo of Monday

Gandoo, 2025-03-14:
ciao @GiulioRomualdi e @isorrentino a few minutes ago I checked the phases of the motor and at the moment they are good.
...good work!!

cc @Fabrizio69  @S-Dafarra

GiulioRomualdi, 2025-03-14:
Pin removed

GiulioRomualdi, 2025-03-14:
Thank you @Gandoo

isorrentino, 2025-03-14:
Thanks a lot, @Gandoo 🚀🚀🚀

closed 2025-03-14 by Fabrizio69