robotology/mesh-support #1983: Unstable ETH communication with the EB3

S-Dafarra, 2024-12-04:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

While attempting to start the robot, we noticed the following errors
```
	739.468796	ERROR			theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb22-j7_10 IP 10.0.1.22 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion
	740.472372	ERROR			theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb22-j7_10 IP 10.0.1.22 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion
	745.456729	ERROR			from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=28s 691m 375u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.
	750.641346	ERROR			[h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 2, 2, .,  , E, x, p, e, c, t, e, d, :,  , 8, 0, 8, 8, ,,  , R, e, c, e, i, v, e, d, :,  , 1, 0, 1, 3, 1, ,,  , M, i, s, s, i, n, g, :,  , 2, 0, 4, 3, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 3, 1, 8, 2, 7, 2, 5, 5,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 3, 3, 8, 7, 1, 2, 5, 5,  , u, s, 
	750.661741	ERROR			from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=33s 891m 375u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.
```

We checked the ping and we noticed that the packet loss was around 15%. We then checked the ping on the board immediately before and we noticed the same issue.

### Detailed context

We launch the following from within the ``ergocub-torso``
```
ping 10.0.1.3 -i 0.002
```
and we obtain the following
```
--- 10.0.1.3 ping statistics ---
3528 packets transmitted, 3169 received, 10.1757% packet loss, time 10013ms
rtt min/avg/max/mdev = 0.258/0.304/0.387/0.010 ms
```

In particular, it is possible to notice some holes like the following
```
64 bytes from 10.0.1.3: icmp_seq=1749 ttl=128 time=0.293 ms
64 bytes from 10.0.1.3: icmp_seq=1750 ttl=128 time=0.315 ms
64 bytes from 10.0.1.3: icmp_seq=1751 ttl=128 time=0.303 ms
64 bytes from 10.0.1.3: icmp_seq=1752 ttl=128 time=0.310 ms
64 bytes from 10.0.1.3: icmp_seq=1753 ttl=128 time=0.309 ms
64 bytes from 10.0.1.3: icmp_seq=1957 ttl=128 time=0.369 ms
64 bytes from 10.0.1.3: icmp_seq=1958 ttl=128 time=0.368 ms
64 bytes from 10.0.1.3: icmp_seq=1959 ttl=128 time=0.352 ms
64 bytes from 10.0.1.3: icmp_seq=1960 ttl=128 time=0.356 ms
64 bytes from 10.0.1.3: icmp_seq=1961 ttl=128 time=0.355 ms
```

### Additional context

We recently updated the firmware of this board together with @traversaro 

cc @Gandoo @filippoborgogni @edersonferrante

Gandoo, 2024-12-05:
ciao, @S-Dafarra @AntonioConsilvio 
checked the cables by doing a board to board bypass. Replaced the shoulder bundle cable but the problem is still there.
I proceed with the replacement of the Board EMS4 EB1

[image]

cc @Fabrizio69 @maggia80

Gandoo, 2024-12-05:
ciao @S-Dafarra @AntonioConsilvio  I replaced the EB1 board and replaced the spacers, below M2.5x5 and between EMS4 and 2Foc M2.5x8 + M2.5x5 to distance the magnetic field.
I replaced the EB3 board because it was damaged, it was losing Eth packets

[image]

[image]

cc @Fabrizio69 @maggia80 @DanielePucci

Gandoo, 2024-12-05:
after the tests carried out by @S-Dafarra  I mounted the covers

 [image]

cc @Fabrizio69 @maggia80 @DanielePucci

AntonioConsilvio, 2024-12-13:
Good job @Gandoo! Closing ✅

closed 2024-12-13 by AntonioConsilvio