robotology/mesh-support #1984: Right front FT sensor is saturated

isorrentino, 2024-12-04:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

[image]

While using the robot the FT started giving these values and saturation errors.

Is it possible to replace it?

### Detailed context

[image]

### Additional context

cc @Gandoo

isorrentino, 2024-12-05:
Cc @Gandoo

Gandoo, 2024-12-05:
ciao @isorrentino @AntonioConsilvio As agreed we changed the two FTSensors from the left shoulder that works with the rear right foot which is broken

[image]

[image]

[image]

cc @filippoborgogni @Fabrizio69 @maggia80 @S-Dafarra @DanielePucci

AntonioConsilvio, 2024-12-17:
Good job! I proceed with the closure of the issue as the problem has been resolved. ✅

We will then follow up with the replacement of the shoulder FT (when we have spare parts available) in an internal issue.

closed 2024-12-17 by AntonioConsilvio